linux-nvme/nvme-cli #414: Reservation Actions Can Overflow into Other Fields

ajeberle, 2018-09-25:
I ran into an issue where during some nvme reservation testing I saw IKEY set after an invalid action value was specified. For example,

`nvme resv-register /dev/nvme0n1 --rrega=10 --nrkey=5`

The rrega action being specified was a typo and should have been 0x2. But the CLI accepts it since it is still less than 1 byte. Since the binary value of 10 is 1010, it sets the RREGA field to 0x2 (010b) and the remaining bit overflows to set the IKEY field.

I looked through the code and found the following in the nvme-ioctl.c file in the nvme_resv_register function:

`__u32 cdw10 = rrega | (iekey ? 1 << 3 : 0) | cptpl << 30;`

Since rrega is not masked out, it follows that if it bit 3 or higher is not zero it can overflow into other fields. A quick check reveals that the same issue is present in nvme_resv_acquire and nvme_resv_release.

I'm not very familiar with the practices in this project or how errors are generally handled. It seems that any changes to the parsing functionality would be overkill and not worth the effort.There could be a check in the resv_register function in nvme.c that returns EINVAL if more than 3 bits are used for the action. Or the actions could be silently masked in the nvme-ioctl.c layer. My gut is to include both, but want a go ahead before I do.

keithbusch, 2018-09-25:
Hey, you're right, thanks for the notice. We definitely want to bounds check everything that has a spec defined range here.  As far as where to validate it, doing both is bonus, but existing checks are closer to the user input, like how cptpl is checked in the same function, so that would be the place I'd recommend.

ajeberle, 2018-09-25:
Just noticed that release and acquire already had the checks in the nvme.c layer for actions, so I'll copy that format, then go ahead and add the extra check in the ioctl layer for all three.

ajeberle, 2018-09-25:
Here is a diff of the changes. It doesn't look like I have permissions to make a pull request:

```
diff --git a/nvme-ioctl.c b/nvme-ioctl.c
index 2a8012c..15d0519 100644
--- a/nvme-ioctl.c
+++ b/nvme-ioctl.c
@@ -267,7 +267,7 @@ int nvme_resv_acquire(int fd, __u32 nsid, __u8 rtype, __u8 racqa,
                      bool iekey, __u64 crkey, __u64 nrkey)
 {
        __le64 payload[2] = { cpu_to_le64(crkey), cpu_to_le64(nrkey) };
-       __u32 cdw10 = racqa | (iekey ? 1 << 3 : 0) | rtype << 8;
+       __u32 cdw10 = (racqa & 0x7) | (iekey ? 1 << 3 : 0) | rtype << 8;
        struct nvme_passthru_cmd cmd = {
                .opcode         = nvme_cmd_resv_acquire,
                .nsid           = nsid,
@@ -283,7 +283,7 @@ int nvme_resv_register(int fd, __u32 nsid, __u8 rrega, __u8 cptpl,
                       bool iekey, __u64 crkey, __u64 nrkey)
 {
        __le64 payload[2] = { cpu_to_le64(crkey), cpu_to_le64(nrkey) };
-       __u32 cdw10 = rrega | (iekey ? 1 << 3 : 0) | cptpl << 30;
+       __u32 cdw10 = (rrega & 0x7) | (iekey ? 1 << 3 : 0) | cptpl << 30;
 
        struct nvme_passthru_cmd cmd = {
                .opcode         = nvme_cmd_resv_register,
@@ -300,7 +300,7 @@ int nvme_resv_release(int fd, __u32 nsid, __u8 rtype, __u8 rrela,
                      bool iekey, __u64 crkey)
 {
        __le64 payload[1] = { cpu_to_le64(crkey) };
-       __u32 cdw10 = rrela | (iekey ? 1 << 3 : 0) | rtype << 8;
+       __u32 cdw10 = (rrela & 0x7) | (iekey ? 1 << 3 : 0) | rtype << 8;
 
        struct nvme_passthru_cmd cmd = {
                .opcode         = nvme_cmd_resv_release,
diff --git a/nvme.c b/nvme.c
index 253bc9e..9e1141a 100644
--- a/nvme.c
+++ b/nvme.c
@@ -3724,6 +3724,12 @@ static int resv_register(int argc, char **argv, struct command *cmd, struct plug
                goto close_fd;
        }
 
+       if (cfg.rrega > 7) {
+               fprintf(stderr, "invalid rrega:%d\n", cfg.rrega);
+               err = EINVAL;
+               goto close_fd;
+       }
+
        err = nvme_resv_register(fd, cfg.namespace_id, cfg.rrega, cfg.cptpl,
                                !!cfg.iekey, cfg.crkey, cfg.nrkey);
        if (err < 0)
```

keithbusch, 2018-09-25:
o pull request permission? Could you format this as a proper patch with your Signed-off-by tag, and post to linux-nvme@lists.infradead.org instead? I have a script to apply them from there.

ajeberle, 2018-09-25:
Thanks for the heads up. Never contributed via mailing list before, so give me a bit to figure out the proper format.

keithbusch, 2018-09-25:
The way I do it:

```
  # git commit -a -s
  # git format-patch HEAD~1 -o ~/
  # git send-email --format-patch ~/0001-<name-of-the-patch> --to=linux-nvme@lists.infradead.org
```

The last part may be a pain to setup (git send-email is not built-in, and setup is specific to your service). So alternatively you could send it as an attachment using your chosen email client, but make sure the email is in plain text as any other format will be rejected.

ajeberle, 2018-09-25:
Appreciate the assistance. I've sent the patch to the mailing list.

keithbusch, 2018-09-25:
i'm afraid it didn't get posted the list. :( If you just want to send it directly, keith.busch@intel.com

ajeberle, 2018-09-26:
Fixed by ade15148bcab98dd32c5718488fbb2f6538105a9

closed 2018-09-26 by ajeberle