linux-nvme/nvme-cli #419: how to find 'lbaf list' using device name?

Gh0stSp1der, 2018-10-11:
Hello, I testing NVMe devices(Intel P4600, Samsung PM1725a, HGST SN100 models) on CentOS 7.4(4.18.4 kernel)

I have a question during testing.
PM1725a & SN100 models are supported multiple namespaces.
As we know, the lbaf list can be found through the namespace ID:
# nvme id-ns /dev/nvme0n1
# nvme id-ns /dev/nvme0 -n1
# smarctl -a /dev/nvme0n1 
(must use the namespace ID for every command.)

However, if you delete the namespace, you can not find the lbaf list through the namespace ID.
So, is there a way to find the lbaf list using the device name(nvmeX) instead of the namespace ID?

If not, I think nvme-cli should support it for the user.
Please answer my question.
Thank you.

keithbusch, 2018-10-11:
The LBAF list is a property of a namespace.

keithbusch, 2018-10-19:
Closing; the request is asking for a feature that doesn't exist in the protocol.

closed 2018-10-19 by keithbusch