linux-nvme/nvme-cli #423: `nvme format` fails with CMD_SEQ_ERROR

zegelin, 2018-10-22:
When trying to run `nvme format` I get the following error:
```
# nvme format /dev/nvme0
NVME Admin command error:CMD_SEQ_ERROR: The command was aborted due to a protocol violation in a multicommand sequence(200c)
```

The device is a M.2 ADATA (XPG) SX8000NP:

```
# nvme id-ctrl /dev/nvme0
NVME Identify Controller:
vid       : 0x126f
ssvid     : 0x126f
sn        : 2I0620030239        
mn        : ADATA SX8000NP                          
fr        : C2.3.7a 
rab       : 6
ieee      : 000000
cmic      : 0
mdts      : 5
cntlid    : 1
ver       : 10200
rtd3r     : 186a0
rtd3e     : 13880
oaes      : 0
ctratt    : 0
rrls      : 0
oacs      : 0x17
acl       : 4
aerl      : 4
frmw      : 0x12
lpa       : 0x3
elpe      : 63
npss      : 4
avscc     : 0x1
apsta     : 0
wctemp    : 335
cctemp    : 345
mtfa      : 50
hmpre     : 0
hmmin     : 0
tnvmcap   : 0
unvmcap   : 0
rpmbs     : 0
edstt     : 5
dsto      : 1
fwug      : 0
kas       : 0
hctma     : 0
mntmt     : 0
mxtmt     : 0
sanicap   : 0
hmminds   : 0
hmmaxd    : 0
nsetidmax : 0
anatt     : 0
anacap    : 0
anagrpmax : 0
nanagrpid : 0
sqes      : 0x66
cqes      : 0x44
maxcmd    : 0
nn        : 1
oncs      : 0x1d
fuses     : 0
fna       : 0x1
vwc       : 0x1
awun      : 4
awupf     : 0
nvscc     : 1
nwpc      : 0
acwu      : 0
sgls      : 0
mnan      : 0
subnqn    : 
ioccsz    : 0
iorcsz    : 0
icdoff    : 0
ctrattr   : 0
msdbd     : 0
ps    0 : mp:9.00W operational enlat:5 exlat:5 rrt:0 rrl:0
          rwt:0 rwl:0 idle_power:- active_power:-
ps    1 : mp:4.60W operational enlat:30 exlat:30 rrt:1 rrl:1
          rwt:1 rwl:1 idle_power:- active_power:-
ps    2 : mp:3.80W operational enlat:30 exlat:30 rrt:2 rrl:2
          rwt:2 rwl:2 idle_power:- active_power:-
ps    3 : mp:0.0700W non-operational enlat:30000 exlat:10000 rrt:3 rrl:3
          rwt:3 rwl:3 idle_power:- active_power:-
ps    4 : mp:0.0050W non-operational enlat:50000 exlat:10000 rrt:4 rrl:4
          rwt:4 rwl:4 idle_power:- active_power:-
```

keithbusch, 2018-10-24:
According to spec, this error should mean your drive is security locked, and you need to unlock it first before running this media altering command. While the spec provides security send/receive commands, it does not define the security protocol so it will be implementation specific and likely require a vendor tool to unlock.

If the device implements a very common security protocol like TCG OPAL, a generic tool like https://github.com/Drive-Trust-Alliance/sedutil might be able to provide more information on its security implementation.

closed 2018-10-25 by keithbusch

zegelin, 2018-10-25:
Ah, good to know. I did try searching for that code but was unable to find anything that pointed towards the cause. Is the NVMe spec available publicly, so that if I see errors in the future I can look them up?

I did find http://nvmexpress.org/wp-content/uploads/NVM-Express-1_3c-2018.05.24-Ratified.pdf, but having a quick read (+ find) through the document I can't see that error code.

keithbusch, 2018-10-25:
Yes, the spec is public and you're linking to the most current public revision (for now).

This error is defined in "Figure 31: Status Code - Generic Command Status Values" in the "0Ch" row.