linux-nvme/nvme-cli #429: Supporting Python 3 in the testsuite

Foxboron, 2018-11-07:
According to the README.md in `tests/` `Python(>= 2.7.5 or >= 3.3)` is supported. But this isn't true. There are a number of tests using `print` instead of `print()` along with basic problems such as bytes and string conversion in python 3. 

The question is whether or not python 2 support should be dropped form the testsuite as it's set to be deprecated in little over a year from now.

```
======================================================================
ERROR: TestNVMeAttachDetachNSCmd (nose2.loader.LoadTestsFailure)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/fox/Git/prosjekter/Python/nvme-cli/.venv/lib/python3.7/site-packages/six.py", line 693, in reraise
    raise value
  File "/home/fox/Git/prosjekter/Python/nvme-cli/.venv/lib/python3.7/site-packages/nose2/plugins/loader/testclasses.py", line 149, in _loadTestsFromTestClass
    for name in names])
  File "/home/fox/Git/prosjekter/Python/nvme-cli/.venv/lib/python3.7/site-packages/nose2/plugins/loader/testclasses.py", line 149, in <listcomp>
    for name in names])
  File "/home/fox/Git/prosjekter/Python/nvme-cli/.venv/lib/python3.7/site-packages/nose2/plugins/loader/testclasses.py", line 193, in __init__
    self.obj = cls()
  File "/home/fox/Git/prosjekter/Python/nvme-cli/tests/nvme_attach_detach_ns_test.py", line 56, in __init__
    self.ctrl_id = self.get_ctrl_id()
  File "/home/fox/Git/prosjekter/Python/nvme-cli/tests/nvme_test.py", line 162, in get_ctrl_id
    ctrl_id = line.split(":")[1].strip()
TypeError: a bytes-like object is required, not 'str'

======================================================================
ERROR: TestNVMeCreateMaxNS (nose2.loader.LoadTestsFailure)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/fox/Git/prosjekter/Python/nvme-cli/.venv/lib/python3.7/site-packages/six.py", line 693, in reraise
    raise value
  File "/home/fox/Git/prosjekter/Python/nvme-cli/.venv/lib/python3.7/site-packages/nose2/plugins/loader/testclasses.py", line 149, in _loadTestsFromTestClass
    for name in names])
  File "/home/fox/Git/prosjekter/Python/nvme-cli/.venv/lib/python3.7/site-packages/nose2/plugins/loader/testclasses.py", line 149, in <listcomp>
    for name in names])
  File "/home/fox/Git/prosjekter/Python/nvme-cli/.venv/lib/python3.7/site-packages/nose2/plugins/loader/testclasses.py", line 193, in __init__
    self.obj = cls()
  File "/home/fox/Git/prosjekter/Python/nvme-cli/tests/nvme_create_max_ns_test.py", line 53, in __init__
    self.nsze = self.get_ncap() / self.get_format() / self.get_max_ns()
  File "/home/fox/Git/prosjekter/Python/nvme-cli/tests/nvme_test.py", line 226, in get_ncap
    if pattern.match(line):
TypeError: cannot use a string pattern on a bytes-like object

======================================================================
ERROR: nvme_error_log_test.TestNVMeErrorLogCmd.test_get_error_log
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/fox/Git/prosjekter/Python/nvme-cli/tests/nvme_error_log_test.py", line 63, in test_get_error_log
    assert_equal(self.get_error_log_ctrl(), 0)
  File "/home/fox/Git/prosjekter/Python/nvme-cli/tests/nvme_error_log_test.py", line 59, in get_error_log_ctrl
    return self.get_error_log()
  File "/home/fox/Git/prosjekter/Python/nvme-cli/tests/nvme_test.py", line 434, in get_error_log
    err_log_entry_count = int(line.split(" ")[5].strip().split(":")[1])
TypeError: a bytes-like object is required, not 'str'

======================================================================
ERROR: TestNVMeFormatCmd (nose2.loader.LoadTestsFailure)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/fox/Git/prosjekter/Python/nvme-cli/.venv/lib/python3.7/site-packages/six.py", line 693, in reraise
    raise value
  File "/home/fox/Git/prosjekter/Python/nvme-cli/.venv/lib/python3.7/site-packages/nose2/plugins/loader/testclasses.py", line 149, in _loadTestsFromTestClass
    for name in names])
  File "/home/fox/Git/prosjekter/Python/nvme-cli/.venv/lib/python3.7/site-packages/nose2/plugins/loader/testclasses.py", line 149, in <listcomp>
    for name in names])
  File "/home/fox/Git/prosjekter/Python/nvme-cli/.venv/lib/python3.7/site-packages/nose2/plugins/loader/testclasses.py", line 193, in __init__
    self.obj = cls()
  File "/home/fox/Git/prosjekter/Python/nvme-cli/tests/nvme_format_test.py", line 68, in __init__
    self.ctrl_id = self.get_ctrl_id()
  File "/home/fox/Git/prosjekter/Python/nvme-cli/tests/nvme_test.py", line 162, in get_ctrl_id
    ctrl_id = line.split(":")[1].strip()
TypeError: a bytes-like object is required, not 'str'

======================================================================
ERROR: nvme_fw_log_test.TestNVMeFwLogCmd.test_fw_log
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/fox/Git/prosjekter/Python/nvme-cli/tests/nvme_fw_log_test.py", line 70, in test_fw_log
    assert_equal(self.get_fw_log(), 0)
  File "/home/fox/Git/prosjekter/Python/nvme-cli/tests/nvme_fw_log_test.py", line 64, in get_fw_log
    print(fw_log_output + "\n")
TypeError: can't concat str to bytes

======================================================================
ERROR: TestNVMeGetMandatoryFeatures (nose2.loader.LoadTestsFailure)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/fox/Git/prosjekter/Python/nvme-cli/.venv/lib/python3.7/site-packages/six.py", line 693, in reraise
    raise value
  File "/home/fox/Git/prosjekter/Python/nvme-cli/.venv/lib/python3.7/site-packages/nose2/plugins/loader/testclasses.py", line 149, in _loadTestsFromTestClass
    for name in names])
  File "/home/fox/Git/prosjekter/Python/nvme-cli/.venv/lib/python3.7/site-packages/nose2/plugins/loader/testclasses.py", line 149, in <listcomp>
    for name in names])
  File "/home/fox/Git/prosjekter/Python/nvme-cli/.venv/lib/python3.7/site-packages/nose2/plugins/loader/testclasses.py", line 193, in __init__
    self.obj = cls()
  File "/home/fox/Git/prosjekter/Python/nvme-cli/tests/nvme_get_features_test.py", line 63, in __init__
    self.vector_list = proc.stdout.read().strip().split(" ")
TypeError: a bytes-like object is required, not 'str'

======================================================================
ERROR: TestNVMeIdentifyNamespace (nose2.loader.LoadTestsFailure)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/fox/Git/prosjekter/Python/nvme-cli/.venv/lib/python3.7/site-packages/six.py", line 693, in reraise
    raise value
  File "/home/fox/Git/prosjekter/Python/nvme-cli/.venv/lib/python3.7/site-packages/nose2/plugins/loader/testclasses.py", line 149, in _loadTestsFromTestClass
    for name in names])
  File "/home/fox/Git/prosjekter/Python/nvme-cli/.venv/lib/python3.7/site-packages/nose2/plugins/loader/testclasses.py", line 149, in <listcomp>
    for name in names])
  File "/home/fox/Git/prosjekter/Python/nvme-cli/.venv/lib/python3.7/site-packages/nose2/plugins/loader/testclasses.py", line 193, in __init__
    self.obj = cls()
  File "/home/fox/Git/prosjekter/Python/nvme-cli/tests/nvme_id_ns_test.py", line 43, in __init__
    self.ns_list = self.get_ns_list()
  File "/home/fox/Git/prosjekter/Python/nvme-cli/tests/nvme_test.py", line 180, in get_ns_list
    ns_list.append(string.replace(line.split(":")[1], '\n', ''))
AttributeError: module 'string' has no attribute 'replace'

======================================================================
ERROR: nvme_smart_log_test.TestNVMeSmartLogCmd.test_smart_log
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/fox/Git/prosjekter/Python/nvme-cli/tests/nvme_smart_log_test.py", line 85, in test_smart_log
    assert_equal(self.get_smart_log_ctrl(), 0)
  File "/home/fox/Git/prosjekter/Python/nvme-cli/tests/nvme_smart_log_test.py", line 60, in get_smart_log_ctrl
    return self.get_smart_log("0xFFFFFFFF")
  File "/home/fox/Git/prosjekter/Python/nvme-cli/tests/nvme_test.py", line 381, in get_smart_log
    if "data_units_read" in line:
TypeError: a bytes-like object is required, not 'str'
```

minwooim, 2019-04-07:
Let's see once PR https://github.com/linux-nvme/nvme-cli/pull/475 is merged.
I have tested it by updating all those test python files with 2to3, it works.

ValZapod, 2020-03-26:
> basic problems such as bytes and string conversion in python 3

So, is that fixed?

closed 2021-04-20 by Foxboron