linux-nvme/nvme-cli #434: Whenever running nvme discover command, syslog shows warning messages

AndyCCChang, 2018-11-27:
Hi,

Whenever running nvme discover command, i.e., nvme discover --transport=rdma --traddr=10.10.0.4, syslog shows the following messages,
Nov 27 15:24:30 initiator-xxx kernel: [55897.822948] nvme nvme1: sqsize 128 > ctrl maxcmd 1, clamping down
Nov 27 15:24:30 initiator-xxx kernel: [55897.822956] nvme nvme1: new ctrl: NQN "nqn.2014-08.org.nvmexpress.discovery", addr 10.10.0.4:4420
Nov 27 15:24:30 initiator-xxx kernel: [55897.823213] nvme nvme1: Removing ctrl: NQN "nqn.2014-08.org.nvmexpress.discovery"
...
I was wondering if there is any way to avoid these warning messages?

OS: Ubuntu 16.04.5
Kernel: 4.18.5-041805-generic
nvme version 1.6.120.g5c8a

Thanks.
Andy

keithbusch, 2018-11-28:
You're talking about the kernel messages, which this tools doesn't control. You may want to take your query to linux-nvme@lists.infradead.org and see if the fabrics developers feel strongly about retaining those messages.

AndyCCChang, 2018-11-29:
Thank you for your reply, appreciate it.

Andy

closed 2018-11-29 by AndyCCChang