linux-nvme/nvme-cli #438: NVME/TCP connection

nickvonklemp, 2018-12-06:
Following commits/merges to [SPDK](https://review.gerrithub.io/c/spdk/spdk/+/425191/) and [NVME host drivers](http://git.infradead.org/nvme.git/commit/eb00c1a1852eb91e1b303aad0cb331318b7b9a0c) I'm trying to connect/mount an nvme drive remotely using tcp instead of soft-roce/rdma. I added an nvmf transport for tcp and a listener for tcp on the server side. 
When trying to connect using the nvme cli on the client I receive the following:
```
# nvme connect-all -t tcp -a <ip_addr>
>> Failed to write to /dev/nvme-fabrics: Invalid argument
```
Dmesg -w also returns:
```
[ 1027.311344] nvme_fabrics: no handler found for transport tcp.
```

Is nvme/tcp unsupported in the nvme cli at this time given how recent the introduced support and merged code is?
Thanks

MiracleMa, 2018-12-27:
I have the same problem with you.
And I can't solve it. So i want to ask whether you have solve it.
QAQ

keithbusch, 2018-12-28:
I may need to get Sagi to help here. Let me catch up on the kernel side implementation in case I took the wrong version into this tree.

danmack, 2019-01-15:
The current version of the nvme cli is working for me with Fabric/TCP:

[dev] # nvme version
nvme version 1.7

[dev] # nvme connect-all -t tcp -a 192.168.190.11
# nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev  
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     CVCQ633400031P2BGN   INTEL SSDPEDMW012T4                      1           1.20  TB /   1.20  TB    512   B +  0 B   8EV10171
/dev/nvme1n1     CVCQ633500RJ1P2BGN   INTEL SSDPEDMW012T4                      1           1.20  TB /   1.20  TB    512   B +  0 B   8EV10171
/dev/nvme2n1     GB000xzxzxzxzxz      PVL-zxzxzxzxzxzx-c0c0c0c0c0             1           2.68  TB /   2.68  TB      4 KiB +  0 B   20018345

I built from master today, rpm/git shows this version:  nvme-1.7.3.geb58-1.el7.x86_64.rpm (eb58f548342f477d02da555cb939b3e6a5609269)

You will need a kernel with matching support and configuration options enabled - here's what we are using if it helps you:

# NVME Support
CONFIG_NVME_CORE=y
CONFIG_BLK_DEV_NVME=y
# CONFIG_NVME_MULTIPATH is not set
CONFIG_NVME_FABRICS=y
# CONFIG_NVME_FC is not set
CONFIG_NVME_TCP=y
# CONFIG_NVME_TARGET is not set
CONFIG_RTC_NVMEM=y
CONFIG_NVMEM=y

sagi-lb, 2019-01-29:
Indeed the NVME_TCP config option needs to be turned on in the kernel build. Once that is in place, the driver should autoload upon nvme connect.

Was this resolved for you?

sagi-lb, 2019-01-29:
are you able to load nvme-tcp driver?

`modprobe nvme-tcp`

damolp, 2020-03-29:
> no handler found for transport tcp

Make sure you're on Kernel v5.0+. Check your config for NVMe TCP (`grep CONFIG_NVME_TCP /boot/config-$(uname -r)`).

You may need the 'extras' kernel module package for your OS (on ubuntu 19.10 / 20.04 this is `linux-modules-extra-$(uname -r)`).

hreinecke, 2022-01-16:
Issue has been answered, closing.

closed 2022-01-16 by hreinecke

lizhihuajob, 2023-10-20:
root@SPDK-SERVER:/home/eht/ceph-nvmeof# modprobe nvme-tcp
modprobe: ERROR: could not insert 'nvme_tcp': Invalid argument

root@SPDK-SERVER:/home/eht/ceph-nvmeof# locate nvme-tcp.ko
/usr/lib/modules/5.15.0-83-generic/kernel/drivers/nvme/host/nvme-tcp.ko
/usr/lib/modules/5.15.0-86-generic/kernel/drivers/nvme/host/nvme-tcp.ko