linux-nvme/nvme-cli #446: nvme: remove duplicate perror in subsystem reset

davidmilburn, 2019-01-09:
# nvme subsystem-reset /dev/nvme0
Subsystem-reset: Inappropriate ioctl for device
Subsystem-reset: NVM Subsystem Reset not supported.

diff --git a/nvme.c b/nvme.c
index 86d8e83..3e74d69 100644
--- a/nvme.c
+++ b/nvme.c
@@ -2670,7 +2670,6 @@ static int subsystem_reset(int argc, char **argv, struct command *cmd, struct pl
        err = nvme_subsystem_reset(fd);
        if (err < 0) {
                close(fd);
-               perror("Subsystem-reset");
                if (errno == ENOTTY)
                        fprintf(stderr,
                                "Subsystem-reset: NVM Subsystem Reset not supported.\n");

davidmilburn, 2019-01-09:
The "perror("Subsystem-reset")" needs to be removed before checking errno.

keithbusch, 2019-01-09:
Right, thanks for the notice. Should be fixed now.

igaw, 2022-01-19:
No progress in this bug report for a long time. If you think it's still relevant please reopen.

closed 2022-01-19 by igaw

vspvijay, 2022-05-28:
This issue is seen on cent os - NVMECLI 1.8 version. Not sure if one can download, compile & install the latest version of NVME CLI.

igaw, 2022-05-30:
Best changes to get the fix ported to Cent OS is to report it to distro bugzilla. I don't expect the Cent OS maintainers monitoring these bugs here.