linux-nvme/nvme-cli #448: [Question] Works in a docker container

Angelinsky7, 2019-01-15:
Hi,
I would like to make this work into a docker container to use it for a prometheus scrapper discovery service.
Usually in that kind of situation we mount the / of the host into /host, or /rootfs of the container.
Could someone tell me what resources (like /sys or /proc) use the cli to access the list and the smart data (at least, like that i could have the temperature and the usage info) like that i could maybe create a PR to add a new command arg that would add a prefix to all resources.

We could that use something like that :
```
nvme --rootfs /host list
nvme --rootfs /host smart-log /dev/nvme0
```

for now if i create a base image with nvme-cli and mount my / into the container with this i got an error
```
docker run --rm -it -v /:/host:ro BASE_IMAGE sh
/ # nvme smart-log /host/dev/nvme0
/host/dev/nvme0: Operation not permitted
```

Thanks for all the help that one could provide
and if someone has a better idea to do that i would gladly hear it !!!

keithbusch, 2019-01-15:
You can get the far majority of functionality with just the admin handle /dev/nvme0.

I'm not sure how to fix this permission error though, and I'm not sure you'd want to. If you could send admin commands to /host/dev/nvme0 from a container, that looks like a security hole. The host owns this character device, not the containers running on the host.

But my container experience is pretty limited. If you want to issue admin commands from a single container, isn't there a way to assign a physical device to one? Something like:

`lxc config device add <container> <name> unix-char path=/dev/nvme0`

?

Angelinsky7, 2019-01-15:
hi @keithbusch and thanks for your answer.
you totally right about the part where the host controls the device but i was naively thinking that if i would want to get information about a device reading was the only "right" needed. So if i would mount (it's not a security hole, i'm allowing it) the device into the container, nvme could read information about it, no ??
i didn't know the possibility that you mentioned about assigning so i'll look at it.

keithbusch, 2019-01-15:
The nvme admin device access is restricted to the root owner only so the driver doesn't need to police what commands are "safe" vs ones that are destructive, or ones that leak information. That decision allows even the oldest driver to have forward compatibility with any vendor specific command and future spec changes.

minwooim, 2019-04-07:
@Angelinsky7 ,
IMHO, if you want to access device node file in host systyem from a container, you can take option:
```
@host         # docker run --rm -it --privileged <image> /bin/bash
@container # nvme smart-log /dev/nvme0
```

Angelinsky7, 2019-04-08:
@minwooim thanks for you answer i'll try that

closed 2022-01-19 by igaw

powerman, 2024-06-15:
Actually `--privileged` is overkill here, it's enough to use `--device /dev/nvme0n1:/dev/nvme0n1` instead.