linux-nvme/nvme-cli #450: The firmware image specified for activation is invalid and not loaded by the controller

tSte, 2019-01-20:
Hey,
I tried to update firmware on X1 Carbon 5 using [this guide](https://gist.github.com/klingtnet/22ab0b907e2d9d20f98c72c93ea5dd37).

Utility successfully loads firmware, but I cannot commit:
```sh
➜ sudo nvme fw-download /dev/nvme0 --fw=NVME_End2End_Lenovo_256_dvt_OPAL_Encrypt_and_Signature.bin
Firmware download success
➜ sudo nvme fw-commit /dev/nvme0 --slot=0 --action=2                                              
NVME Admin command error:FIRMWARE_IMAGE: The firmware image specified for activation is invalid and not loaded by the controller(107)
```

I tried latest version (`fwnv28`), first version that included this drive (`fwnv16`) and even version that was used in that guide (`fwnv25`). I tried both `--action=1` and `2`. Out of curiosity, I even tried both `256` and `512`.

My drive is `INTEL SSDPEKKF256G7L` and it is included in supported drives in [readme](https://download.lenovo.com/pccbbs/mobiles/fwnv28en.txt) so I should have proper binary. My current firmware version is `118P`.

Do you guys have any idea what could be wrong? Anything...
Many thanks.

keithbusch, 2019-01-22:
I think you want --action=1. Action '2' says to activate the firmware in --slot, which you're saying is '0', which isn't a valid slot (it's a directive to the controller to pick a slot, but that directive works with only certain actions).

But you say you tried action=1 ... So the error seems to indicate the running firmware doesn't believe it is compatible with the requested firmware, or it doesn't recognize the signature.

tSte, 2019-01-23:
Thanks for reply.
@keithbusch I don't think that I am using wrong binary (I checked supported drives), but regarding that signature... with what signature does it compare with?

keithbusch, 2019-01-23:
How firmware validates other firmware is a vendor and oem specific function. They'll typically digitally sign their binaries and reject ones that they can't decode. I'd have to recommend escalating this toward the OEM if their instructions are not working with their provided binaries.

tSte, 2019-01-23:
Okay, thanks

closed 2019-01-23 by tSte