linux-nvme/nvme-cli #454: insufficiant capacity

scheuref, 2019-01-31:
Dear Maintainers,

We tried to replace the default namespace of 8 TB on a INTEL SSDPE2KX080T8 with 2 namespaces of 4 TB.
We had several strange issue with recent kernels.

On SLES12 SP3 (4.4.162-94.72-default #1 SMP Mon Nov 12 18:57:45 UTC 2018 (9de753f) x86_64 x86_64 x86_64 GNU/Linux)
with nvme-cli v1.2 we could delete the ns but we could create one ns. This seems to be a limitation from intel.
However, recreating the ns with the original size gave a NS_INSUFFICIENT_CAPACITY error,
but trying with a size 64 MiB smaller was successful.

On recent kernels, for ex. grml 4.19.0-1-grml-amd64 #1 SMP Debian 4.19.8-1+grml.1 (2018-12-11) x86_64 GNU/Linux
with nvme-cli v1.6 we had the same issue with lower capacity after deleting a ns.
And we also had a much serious bug: nvme-cli was sending the commands to the wrong nvme's.
For ex., a format command on nvme0 was wiping all data on nvme3 !!! (fortunately it was a test server...)
We build nvme-cli from the current master branch (v1.7.18.g637b) with the same results.

1st Bug - insufficiant capacity:
    #Note: /dev/nvme4 is here the controller of /dev/nvme1n1 BTLJ841002H88P0HGN, see description of second bug.

    root@grml:~/nvme-cli# nvme create-ns /dev/nvme4 --nsze=$((0x3a3812ab0)) --ncap=$((0x3a3812ab0)) --flbas=0 --dps=0
        NVMe Status:NS_INSUFFICIENT_CAPACITY: Creating the namespace requires more free space than is currently available. The Command Specific Information field of the Error Information Log specifies the total amount of NVM capacity required to create the namespace in bytes(4115)

    root@grml:~/nvme-cli# nvme create-ns /dev/nvme4 --nsze=$((0x3a3812ab0-1024*128)) --ncap=$((0x3a3812ab0-1024*128)) --flbas=0 --dps=0
        create-ns: Success, created nsid:1

    #left: original (untouched) ns of 0x3a3812ab0 sectors
    #right: recreated ns of 0x3a3812ab0 - 1024*128 sectors, with a lost capacity of 64 MiB

    root@grml:~# sdiff <(nvme id-ns /dev/nvme0n1) <(nvme id-ns /dev/nvme1n1)
        NVME Identify Namespace 1:                  NVME Identify Namespace 1:
        nsze    : 0x3a3812ab0                         | nsze    : 0x3a3800000
        ncap    : 0x3a3812ab0                         | ncap    : 0x3a3800000
        nuse    : 0x3a3812ab0                         | nuse    : 0x3a3800000
        nsfeat  : 0                         nsfeat  : 0
        nlbaf   : 1                         nlbaf   : 1
        flbas   : 0                         flbas   : 0
        mc      : 0                         mc      : 0
        dpc     : 0                         dpc     : 0
        dps     : 0                         dps     : 0
        nmic    : 0                         nmic    : 0
        rescap  : 0                         rescap  : 0
        fpi     : 0                         fpi     : 0
        dlfeat  : 0                         dlfeat  : 0
        nawun   : 0                         nawun   : 0
        nawupf  : 0                         nawupf  : 0
        nacwu   : 0                         nacwu   : 0
        nabsn   : 0                         nabsn   : 0
        nabo    : 0                         nabo    : 0
        nabspf  : 0                         nabspf  : 0
        noiob   : 0                         noiob   : 0
        nvmcap  : 8001563222016                       | nvmcap  : 8001524072448
        nvmsetid: 0                         nvmsetid: 0
        endgid  : 0                         endgid  : 0
        nguid   : 01000000010000005cd2e4cc48005051            | nguid   : 01000000080000005cd2e413b0005051
        eui64   : 5cd2e4cc48000100                    | eui64   : 5cd2e413b0000800
        lbaf  0 : ms:0   lbads:9  rp:0x2 (in use)           lbaf  0 : ms:0   lbads:9  rp:0x2 (in use)

Best Regards
Francois Scheurer

scheuref, 2019-01-31:
linked to issue [455](https://github.com/linux-nvme/nvme-cli/issues/455)

keithbusch, 2019-01-31:
Sounds like a firmware issue to me. Their create namespace implementation is probably rounding up to some granularity (say, 1GB).

closed 2019-03-18 by keithbusch