linux-nvme/nvme-cli #458: LBA format error on Samsung PM981 and SM1715

111alan, 2019-02-23:
On SM1715 and PM981, the default LBA format is 0(512B+0B). I can't do
`sudo nvme format /dev/nvme0n1`
But I can do
`sudo nvme format /dev/nvme0n1 --lbaf=1(or 3 but not 2)`
But after that there's no way to return to 0(512+0) or 2(4K+0), so it's stuck to the formats with metadata, 
`sudo nvme format /dev/nvme0n1 --lbaf=0(or 2)`
still won't work, they both return"the LBA format specified is not supported. This may be due to various conditions(50a)".

Are there any workaround for this? Or it's a bug in nvme-cli, or a proprietary plugin is needed?

Thanks.

[image]

keithbusch, 2019-02-25:
Looks like a device side bug. Should report to the vendor.

visheshvijay, 2019-03-25:
I guess to overcome this you may need to delete the NS and recreate it agin as per your desired format.
Most vendors dont allow from 512 to 512+8 or 4K to 4K+8.
This is due to backend firmware limitation.

closed 2022-01-19 by igaw