linux-nvme/nvme-cli #462: nvme cmd could not discover more than 3 targets

krishna-e, 2019-03-17:
The below nvme command could not discover more than 3 targets(say the target machine has 8 nvme targets).

	#nvme discover -t rdma -a 102.1.1.6 -s 4420
	
Note that the 4th record shows "unrecognized" for trtype & subtype:

			Discovery Log Number of Records 8, Generation counter 8
			=====Discovery Log Entry 0======
			trtype:  rdma
			adrfam:  ipv4
			subtype: nvme subsystem
			treq:    not specified
			portid:  1
			trsvcid: 4420
			subnqn:  nvme-ram0
			traddr:  102.1.1.6
			rdma_prtype: not specified
			rdma_qptype: connected
			rdma_cms:    rdma-cm
			rdma_pkey: 0x0000
			=====Discovery Log Entry 1======
			trtype:  rdma
			adrfam:  ipv4
			subtype: nvme subsystem
			treq:    not specified
			portid:  1
			trsvcid: 4420
			subnqn:  nvme-ram1
			traddr:  102.1.1.6
			rdma_prtype: not specified
			rdma_qptype: connected
			rdma_cms:    rdma-cm
			rdma_pkey: 0x0000
			=====Discovery Log Entry 2======
			trtype:  rdma
			adrfam:  ipv4
			subtype: nvme subsystem
			treq:    not specified
			portid:  1
			trsvcid: 4420
			subnqn:  nvme-ram2
			traddr:  102.1.1.6
			rdma_prtype: not specified
			rdma_qptype: connected
			rdma_cms:    rdma-cm
			rdma_pkey: 0x0000
			=====Discovery Log Entry 3======
			trtype:  unrecognized
			adrfam:  pci
			subtype: unrecognized
			treq:    not specified
			portid:  0
			trsvcid:
			subnqn:
			traddr:

This issue is only observed with v1.7 and TOT.
And with v1.6 I'm able to discover more than 3 targets without any issue. 

Suspecting the commit 465a4d5, I reverted the code changes and then the discover works fine.

		https://github.com/linux-nvme/nvme-cli/commit/465a4d5ab209bb20ea4b3ba808438747773edd2b
		
Can someone comment if this issue is really due to above commit or not?

===============Full output of discover command=========================
# ./nvme discover -t rdma -a 102.1.1.6 -s 4420

Discovery Log Number of Records 8, Generation counter 8
=====Discovery Log Entry 0======
trtype:  rdma
adrfam:  ipv4
subtype: nvme subsystem
treq:    not specified
portid:  1
trsvcid: 4420
subnqn:  nvme-ram0
traddr:  102.1.1.6
rdma_prtype: not specified
rdma_qptype: connected
rdma_cms:    rdma-cm
rdma_pkey: 0x0000
=====Discovery Log Entry 1======
trtype:  rdma
adrfam:  ipv4
subtype: nvme subsystem
treq:    not specified
portid:  1
trsvcid: 4420
subnqn:  nvme-ram1
traddr:  102.1.1.6
rdma_prtype: not specified
rdma_qptype: connected
rdma_cms:    rdma-cm
rdma_pkey: 0x0000
=====Discovery Log Entry 2======
trtype:  rdma
adrfam:  ipv4
subtype: nvme subsystem
treq:    not specified
portid:  1
trsvcid: 4420
subnqn:  nvme-ram2
traddr:  102.1.1.6
rdma_prtype: not specified
rdma_qptype: connected
rdma_cms:    rdma-cm
rdma_pkey: 0x0000
=====Discovery Log Entry 3======
trtype:  unrecognized
adrfam:  pci
subtype: unrecognized
treq:    not specified
portid:  0
trsvcid:
subnqn:
traddr:
=====Discovery Log Entry 4======
trtype:  rdma
adrfam:  ipv4
subtype: nvme subsystem
treq:    not specified
portid:  1
trsvcid: 4420
subnqn:  nvme-ram0
traddr:  102.1.1.6
rdma_prtype: not specified
rdma_qptype: connected
rdma_cms:    rdma-cm
rdma_pkey: 0x0000
=====Discovery Log Entry 5======
trtype:  rdma
adrfam:  ipv4
subtype: nvme subsystem
treq:    not specified
portid:  1
trsvcid: 4420
subnqn:  nvme-ram1
traddr:  102.1.1.6
rdma_prtype: not specified
rdma_qptype: connected
rdma_cms:    rdma-cm
rdma_pkey: 0x0000
=====Discovery Log Entry 6======
trtype:  rdma
adrfam:  ipv4
subtype: nvme subsystem
treq:    not specified
portid:  1
trsvcid: 4420
subnqn:  nvme-ram2
traddr:  102.1.1.6
rdma_prtype: not specified
rdma_qptype: connected
rdma_cms:    rdma-cm
rdma_pkey: 0x0000
=====Discovery Log Entry 7======
trtype:  unrecognized
adrfam:  pci
subtype: unrecognized
treq:    not specified
portid:  0
trsvcid:
subnqn:
traddr:

Thanks,
Krishna.

keithbusch, 2019-03-18:
Interesting that commit harmed this. It was meant to fix MDTS issues that could only respond with partial log pages. Is the nvme target a linux target?

krishna-e, 2019-03-18:
> 
> 
> Interesting that commit harmed this. It was meant to fix MDTS issues that could only respond with partial log pages. Is the nvme target a linux target?

Yes, it's a Linux target.

# uname -a
Linux localhost.localdomain 4.19.15 #1 SMP Fri Mar 15 16:58:42 IST 2019 x86_64 x86_64 x86_64 GNU/Linux
# cat /etc/redhat-release
Red Hat Enterprise Linux Server release 7.5 (Maipo)

krishna-e, 2019-03-20:
> 
> 
> Interesting that commit harmed this. It was meant to fix MDTS issues that could only respond with partial log pages. Is the nvme target a linux target?

Tried on linux 5.0.0 also, observed the same issue.
(both host and target have linux 5.0.0 )

    #uname -a
    Linux localhost.localdomain 5.0.0 #1 SMP Wed Mar 20 19:06:59 IST 2019 x86_64 x86_64 x86_64 GNU/Linux

    #cat /etc/redhat-release
    Red Hat Enterprise Linux Server release 7.5 (Maipo)

krishna-e, 2019-03-26:
Any thoughts on this, please?

Kajtek, 2019-03-29:
I have this same issue with 1.7 version. I did additional experiments and noticed that only 4th record is malformed.
When I retrieved 10 records, only 4th was corrupted.

minwooim, 2019-03-31:
The reason why 4th entry is broken is that based on linux-5.0, nvmet_execute_get_disc_log_page() in drivers/nvme/target/discovery.c deos not support offset for GET LOG PAGE with LPOL and LPOU(Log Page Offset Lower/Upper).

The first three entries with header will be 4096bytes so that the first data transfer whose offset is 0x0 seems going without any issue.
- sizeof(Discovery Log Page Entry) == 1024
- sizeof(Discovery Log Page Header) == 1024

The second GET LOG PAGE command will receive data from offset 0x0 due to not considering offset in nvmet discovery.

keithbusch, 2019-04-01:
Thanks, @minwooim . This sounds like an issue we should take to linux-nvme mailing list.

keithbusch, 2019-04-08:
The kernel fix is posted here: http://lists.infradead.org/pipermail/linux-nvme/2019-April/023293.html

If the others are okay with this, I'll see if we can get this ported to linux-stable.

I may also need to add a parameter to nvme-cli's discover to take a host buffer request so we can work-around this list corruption.

minwooim, 2019-04-09:
@keithbusch ,
I think that looks good to me.  I have no idea why I have not received that mail while I'm surely subscribing linux-nvme mailing list. Anyway If it's merged to to stable, It would be better.

Thanks,

keithbusch, 2019-04-10:
The v3 patch was applied for a 5.1 merge, and should hit the rc5. The fix unfortunately is not a cherry for any of the applicable LTS kernels...

Shubham1995pardeshi, 2019-04-24:
I am also hitting this issue, and using kernel version 5.0.0-13-generic Ubuntu 19.04
If the fix is ready then can we cut new release ?

Thanks,
Shubham Pardeshi

Shubham1995pardeshi, 2019-04-24:
I have compiled and installed released nvme cli version 1.8.1

minwooim, 2019-04-24:
@Shubham1995pardeshi ,
The fix has been merged to v5.1-rc5 tag:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?h=v5.1-rc5&id=d808b7f759b50acf0784ce6230ffa63e12ef465d

The kernel version that you are using now might not contain the fix from
Keith.

krishna-e, 2019-05-09:
Thanks for fixing the issue.  I verified the fix on 5.1.0-rc6.
Closing as the issue got resolved.

keithbusch, 2019-05-09:
This was also picked up for 5.0's stable queue. The 4.19 LTS would have been even better, but that's going to require a back-port.