linux-nvme/nvme-cli #463: NVMe Entry & Exit Latency

acharyasunil, 2019-03-18:
Hope to get help on this.

I am trying to get Entry & Exit Latency when manually switched from one power state to another power state ( Either from Active to Active OR Active to Idle OR Vice Versa Power state ).

Upon executing NVMe Identify Controller command:
`nvme id-ctrl /dev/nvme0n1 `
i could see the list of power states at the end, according to my understanding the Entry and the Exit Latency values specified are the maximum latency but i need the actual entry and exit latency when i manually switch from one power state to another power state.

Is it possible that Maximum latency that i am referring to is same as that of the actual latency that i am trying to find?

Any help regarding this is appreciated.
Thanks!

keithbusch, 2019-03-18:
The nvme spec only provides the maximum. There is no spec method to report the operational latency of a previous power state transition.

acharyasunil, 2019-03-20:
Thanks for the reply @keithbusch 

Understood that NVMe spec doesn't provide operational latency.

So i am now Exploring on PCIe Power state when connected to NVMe drive instead. 

I am trying to measure PCIe Bus Link L0, L0s, L1 power states Entry & Exit Latency.

i tried lspci -vvv to get insights. it shows L0 Unlimited, L1 < 64ms etc. such values.
How to get operational latency when we switch between PCIe L0 & L1 states?

May be this question is off topic but hope to get some more insights on this
Thanks again

keithbusch, 2019-03-20:
just the max. you'd need to instrument something to measure it.

closed 2022-01-19 by igaw