linux-nvme/nvme-cli #481: reset after firmware update

vbraun, 2019-04-22:
I successfully updated my PM961 to 5L7QCXB7 using `nvme`, but I had to reboot my computer until `nvme id-ctrl /dev/nvme0` showed the new version. The documented `echo 1 > /sys/class/nvme/nvme0/device/reset` did not work, that is, `nvme id-ctrl /dev/nvme0` still showed the old version. 

Perhaps the `nvme-fw-commit` man page should just instruct users to reboot instead? 
~~~~
       No further action is automatically taken to reset the device, which is usually required to complete the activation process. If your kernel and
       driver are recent enough, you can commit the firmware by issuing a reset through Linux sysfs, for example:

            # echo 1 > /sys/class/nvme/nvme0/device/reset

       If your kernel is not recent enough, you will need to remove and add the device some other way.
~~~~
My kernel 5.0.7-200.fc29.x86_64 is newer than the man page so presumably it is "recent enough"...

keithbusch, 2019-04-22:
Yes, that version does have the capabilities.

The man page is a little incomplete, though. The spec allows a firmware update to take affect in one of several ways. Most controllers support activating new firmware from an NVMe level reset (echo 1 > /sys/class/nvme/nvme0/reset_controller), and others may require a PCI function level reset, which is what the man-page shows.

It sounds like, though, your device requires  a PCI cold reset, and we currently don't have a good way to support that.

vbraun, 2019-04-22:
Thanks for the info! I think it would be great if that paragraph could be added to the man page.

For  the record, this is a Samsung PM961 on an Asus crosshair VI hero AM4 mainboard.

closed 2022-01-19 by igaw