linux-nvme/nvme-cli #482: Compile failed on Ubuntu 19.04

Shubham1995pardeshi, 2019-04-23:
```

make 
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1"' -o argconfig.o -c argconfig.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1"' -o suffix.o -c suffix.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1"' -o parser.o -c parser.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1"' -o nvme-print.o -c nvme-print.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1"' -o nvme-ioctl.o -c nvme-ioctl.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1"' -o nvme-lightnvm.o -c nvme-lightnvm.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1"' -o fabrics.o -c fabrics.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1"' -o json.o -c json.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1"' -o nvme-models.o -c nvme-models.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1"' -o plugin.o -c plugin.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1"' -o plugins/intel/intel-nvme.o -c plugins/intel/intel-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1"' -o plugins/lnvm/lnvm-nvme.o -c plugins/lnvm/lnvm-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1"' -o plugins/memblaze/memblaze-nvme.o -c plugins/memblaze/memblaze-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1"' -o plugins/wdc/wdc-nvme.o -c plugins/wdc/wdc-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1"' -o plugins/wdc/wdc-utils.o -c plugins/wdc/wdc-utils.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1"' -o plugins/huawei/huawei-nvme.o -c plugins/huawei/huawei-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1"' -o plugins/netapp/netapp-nvme.o -c plugins/netapp/netapp-nvme.c
plugins/netapp/netapp-nvme.c: In function 'netapp_ontapdevices':
plugins/netapp/netapp-nvme.c:196:44: error: '%s' directive output may be truncated writing up to 259 bytes into a region of size between 255 and 514 [-Werror=format-truncation=]
  snprintf(nspath, ONTAP_NS_PATHLEN, "%s%s%s%s", ontap_vol,
                                            ^~
    vol_name, "/", ns_name);
                   ~~~~~~~                   
In file included from /usr/include/stdio.h:867,
                 from plugins/netapp/netapp-nvme.c:16:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:67:10: note: '__builtin___snprintf_chk' output between 7 and 525 bytes into a destination of size 520
   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        __bos (__s), __fmt, __va_arg_pack ());
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [Makefile:59: plugins/netapp/netapp-nvme.o] Error 1

```

keithbusch, 2019-04-23:
Do you have this commit? https://github.com/linux-nvme/nvme-cli/commit/fefedba848578c83ccefcb9be0b71f13e9913481

Shubham1995pardeshi, 2019-04-23:
root@mankad:~# cat /etc/os-release 
NAME="Ubuntu"
VERSION="19.04 (Disco Dingo)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 19.04"
VERSION_ID="19.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=disco
UBUNTU_CODENAME=disco
root@mankad:~# uname -r
**5.0.0-13-generic**
root@mankad:~# gcc --version
**gcc (Ubuntu 8.3.0-6ubuntu1) 8.3.0**
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

root@mankad:~# make --version
**GNU Make 4.2.1**
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
root@mankad:~# nvme --version
**nvme version 1.8.1**
root@mankad:~#

keithbusch, 2019-04-23:
I see your installed version is 1.8.1, but what version are you trying to compile?

keithbusch, 2019-04-23:
What is the output of `git log --oneline -n 1`?

Shubham1995pardeshi, 2019-04-23:
This is released 1.8.1 version.

keithbusch, 2019-04-23:
ok, you'll need the fix I pointed to: https://github.com/linux-nvme/nvme-cli/commit/fefedba848578c83ccefcb9be0b71f13e9913481

Shubham1995pardeshi, 2019-04-23:
Yes, I want fix. Can you please release new version. And close this issue.

Thanks,
Shubham Pardeshi

keithbusch, 2019-04-23:
Fix is in repository, if you want to grab that. If you just want to compile the most recent tag, you can override the harmless warning, `CFLAGS="-O2 -g" make`.

I'll need to accumulate some more patches before making a new tag.

closed 2022-01-19 by igaw