linux-nvme/nvme-cli #484: nvme 1.8.1 : connect-all command returned warning "skipping unsupported subtype 0" But volumes are connected.

Shubham1995pardeshi, 2019-04-24:
```
root@mankad:~# nvme connect-all -t tcp -i 8 -s 4420 -a 192.168.40.31
skipping unsupported subtype 0
root@mankad:~# 
```

Shubham1995pardeshi, 2019-04-24:
```
root@mankad:~# uname -r
5.0.0-13-generic
root@mankad:~# nvme --version
nvme version 1.8.1
root@mankad:~# cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=19.04
DISTRIB_CODENAME=disco
DISTRIB_DESCRIPTION="Ubuntu 19.04"
root@mankad:~# 

```

minwooim, 2019-04-24:
Your Kernel(5.0.0-) seems not to have the fix for the offset-given discovery process.  That might cause the unsupported subtype error during trying to connect to the controller.

You can find out the fix at (5.1-rc5)
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?h=v5.1-rc5&id=d808b7f759b50acf0784ce6230ffa63e12ef465d

keithbusch, 2019-04-24:
5.0 does not have the fix. 5.1 will be the first to see it, and it unfortunately does not cleanly port to any stable release. I'm almost inclined to create or accept a user-space work-around for this issue.

sagi-lb, 2019-04-24:
Hey Keith, do we have a choice?

keithbusch, 2019-04-24:
i suppose not ... \_(x_x)_/