linux-nvme/nvme-cli #486: nvme options to print without headers

bajrang0789, 2019-04-25:
Is there any option to print the list output from nvme without the headers.

I am trying to assign a device drive for an NVMe Storage on our AWS EC2 instance. 
1.) 
`ubuntu@ip:~/ansible$ sudo nvme list
`
Output:

```
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     vol08ea1f8c844a0ad7a Amazon Elastic Block Store               1           0.00   B /  21.47  GB    512   B +  0 B   1.0
/dev/nvme1n1     vol0aa5ede7637c43aa6 Amazon Elastic Block Store               1           0.00   B /   8.59  GB    512   B +  0 B   1.0

```
Expected / Required Op: (without the headers)
```
/dev/nvme0n1     vol08ea1f8c844a0ad7a Amazon Elastic Block Store               1           0.00   B /  21.47  GB    512   B +  0 B   1.0
/dev/nvme1n1     vol0aa5ede7637c43aa6 Amazon Elastic Block Store               1           0.00   B /   8.59  GB    512   B +  0 B   1.0
```
2.) 
```
ubuntu@ip:~/ansible$ sudo nvme id-ctrl --raw-binary /dev/nvme0n1  | cut -c3073-3104 | tr -s ' ' | sed 's/ $//g'
```

The output is : 
```
binary output
xvda
```

I just need `xvda` as the output header

bajrang0789, 2019-04-25:
nvme version 1.5
OS: Ubuntu 18.04

minwooim, 2019-04-25:
I'm not sure this is what you really want to do, but you can try with the following bash.

```bash
# nvme list | tail -n+3
```

keithbusch, 2019-04-25:
yeah, @minwooim suggestion is the recommended way to manipulate the output. There's also the 'json' format that may be easier for a program to parse.

keithbusch, 2019-08-30:
Hey, so on the "binary output", that's printed to stderr. The part you want is sent to stdout, so these can be piped separately, redirected to whichever consumer you want for it.

That said, I'm not sure why it's printed at all.

pkit, 2021-04-02:
@bajrang0789 
```
$ sudo nvme list -o json | jq .Devices[].DevicePath -r
/dev/nvme0n1
/dev/nvme1n1
```

or the full solution for that AWS braindead problem
```
$ sudo nvme list -o json | jq .Devices[].DevicePath -r | xargs -i bash -c "echo {} && sudo nvme id-ctrl -v {} | grep -oP '^0000:.* \"\K[^.]+'" | sed 'N;s/\n/ /'
/dev/nvme0n1 sda1
/dev/nvme1n1 /dev/sdf
```

pkit, 2021-04-02:
@keithbusch on an related note, why the vendor specific data is **not** printed in `-o json` ?
Just curious.

keithbusch, 2021-04-02:
Without a vendor extension/plugin, we don't know how to parse the vendor specific data. All we could do is report it has a hex dump.

pkit, 2021-04-02:
But essentially that dump is unavailable in json output...

keithbusch, 2021-04-02:
The amazon addition decodes the vendor specific region from identify controller. Is that the only structure you're interested in? It's included in the 'amzn id-ctrl' output already.

closed 2022-01-19 by igaw