linux-nvme/nvme-cli #495: asciidoctor 2.x - no more docboox45 support

DimStar77, 2019-05-08:
https://github.com/linux-nvme/nvme-cli/blob/17da35105ba04c1cc22198aba74f23b8a6fac33d/Documentation/Makefile#L43

It is not possible to build the documentation using asciidoctor 2.x as the docboox45 backend has been dropped Keep in mind that docboox5 apepeared like > 10 years ago, so moving away from docboox45 to docbook5 sounds really sensible.

keithbusch, 2019-05-08:
Sure, I guess that's fine, though I've never built this with asciidoctor before, and it doesn't appear to validate currently either way. This part is just copied straight out of the git git repository: https://github.com/git/git/blob/master/Documentation/Makefile#L192

DimStar77, 2019-05-08:
[nvme-cli-docbook5.patch.txt](https://github.com/linux-nvme/nvme-cli/files/3158307/nvme-cli-docbook5.patch.txt)

This is what I tried to far - and the resutls seem reasonable so far (xmlto and docbook5 is also no est friend)

keithbusch, 2019-05-08:
Looks fine to me. Want to send as a patch with change log and sign-off? Either pull request or email to the linux nvme mailing list is fine.

PR #496 (2019-05-08): DOC: Move to DocBook 5 when using asciidoctor
DocBook 5 has been released about a decade ago. Newer versions of
asciidoctor (since version 2.x) no longer support the previously
used DocBook 4.x format, so it is about time for us to switch too.

Fixes #495
commit c2ef75ce (2019-05-08): DOC: Move to DocBook 5 when using asciidoctor

DocBook 5 has been released about a decade ago. Newer versions of
asciidoctor (since version 2.x) no longer support the previously
used DocBook 4.x format, so it is about time for us to switch too.

Fixes #495
closed 2019-05-08 by keithbusch