linux-nvme/nvme-cli #499: disconnect using device doesn't remove all controllers, further nvme list gives errors

Shubham1995pardeshi, 2019-05-14:
**Steps followed**
```

root@mankad:~# nvme list
root@mankad:~# nvme connect -t tcp -a 192.168.140.231 -s 4420 -n GB00002004bbf97 -i 8
root@mankad:~# nvme connect -t tcp -a 192.168.140.141 -s 4420 -n GB00002004bbf97 -i 8
root@mankad:~# nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev  
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     GB00002004bbf97      PVL-MX18S0P2L2C1-F100TP0TY1              1         107.37  GB / 107.37  GB      4 KiB +  0 B   22119177
root@mankad:~# nvme disconnect -d nvme0
root@mankad:~# nvme list
/dev/nvme0n1: failed to obtain nvme info: Operation not permitted
root@mankad:~# ls -ltra /dev/ | grep nvme
crw-------  1 root root     10,  54 May 13 23:16 nvme-fabrics
brw-rw----  1 root disk    259,   2 May 13 23:18 nvme0n1
crw-------  1 root root    236,   1 May 13 23:18 nvme1
root@mankad:~# 
```

**man page info for nvme disconnect**
```

NVME-DISCONNECT(1)                                                                            NVMe Manual                                                                            NVME-DISCONNECT(1)

NAME
       nvme-disconnect - Disconnect one or more Fabrics controller(s).

SYNOPSIS
       nvme disconnect
                       [--nqn=<subnqn>           | -n <subnqn>]
                       [--device=<device>        | -d <device>]

DESCRIPTION
       Disconnects and removes one or more existing NVMe over Fabrics controllers. If the --nqn option is specified all controllers connecting to the Subsystem identified by subnqn will be removed.
       If the --device option is specified the controller specified by the --device option will be removed.

OPTIONS
       -n <subnqn>, --nqn <subnqn>
           Indicates that all controllers for the NVMe subsystems specified should be removed.

       -d <device>, --device <device>
           Indicates that the controller with the specified name should be removed.

EXAMPLES
       o   Disconnect all controllers for a subsystem named nqn.2014-08.com.example:nvme:nvm-subsystem-sn-d78432:

               # nvme disconnect --nqn=nqn.2014-08.com.example:nvme:nvm-subsystem-sn-d78432

       o   Disconnect the controller nvme4

               # nvme disconnect --device=nvme4

```

keithbusch, 2019-05-14:
This looks like a kernel bug.

minwooim, 2019-05-14:
Now I can reproduce this issue in my env.  It seems like the issue is in case name
of subsystem in different targets are the same.  If I figure out the root cause of this
issue before somebody else figure it out what happened, I will share on this issue.

minwooim, 2019-05-14:
I have tried the same thing here:
```
root@host:~# nvme connect -t tcp -a 192.168.0.13 -s 4420 -n GB00002004bbf97
root@host:~# nvme connect -t tcp -a 192.168.0.15 -s 4420 -n GB00002004bbf97
```
And kernel says like:
```
[   57.244072] nvme nvme1: creating 8 I/O queues.                                 
[   57.284495] nvme nvme1: new ctrl: NQN "GB00002004bbf97", addr 192.168.0.13:4420
[   65.315385] nvme nvme2: creating 8 I/O queues.                                 
[   65.355146] nvme nvme2: new ctrl: NQN "GB00002004bbf97", addr 192.168.0.15:4420
```

Those two controller should be initialized in my env with nvme1 and nvme2.
I can't see namespace has been initialized for nvme2 controller:
```
nvme1  
nvme1n1
nvme2  
```
I think disconnect might not be the issue here.  The connect seems
not proceeded properly.

keithbusch, 2019-05-14:
Probably multipath.

minwooim, 2019-05-14:
Yeah, I missed that one.  Thanks Keith.

closed 2022-01-19 by igaw