linux-nvme/nvme-cli #500: NVMe Status:INVALID_LOG_PAGE

BenoitPel, 2019-05-14:
Hi,
I have install this package : nvme-cli_1.7-1_amd64.deb

I try show endurance-log on my nvme device : 
`nvme endurance-log /dev/nvme1
NVMe Status:INVALID_LOG_PAGE: The log page indicated is invalid. This error condition is also returned if a reserved log page is requested(4109)`
`nvme endurance-log /dev/nvme1n1 
NVMe Status:INVALID_LOG_PAGE: The log page indicated is invalid. This error condition is also returned if a reserved log page is requested(4109)`
`nvme endurance-log /dev/nvme0
NVMe Status:INVALID_LOG_PAGE: The log page indicated is invalid. This error condition is also returned if a reserved log page is requested(4109)`
`nvme endurance-log /dev/nvme0n1 
NVMe Status:INVALID_LOG_PAGE: The log page indicated is invalid. This error condition is also returned if a reserved log page is requested(4109)`
`smartctl -x /dev/nvme1
smartctl 6.6 2016-05-31 r4324 [x86_64-linux-4.9.0-9-amd64] (local build)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Number:                       WDC CL SN720 SDAQNTW-512G-2000
Serial Number:                      1851BC802753
Firmware Version:                   10104122
PCI Vendor/Subsystem ID:            0x15b7
IEEE OUI Identifier:                0x001b44
Total NVM Capacity:                 512 110 190 592 [512 GB]
Unallocated NVM Capacity:           0
Controller ID:                      8215
Number of Namespaces:               1
Namespace 1 Size/Capacity:          512 110 190 592 [512 GB]
Namespace 1 Formatted LBA Size:     512
Local Time is:                      Tue May 14 11:02:35 2019 CEST
Firmware Updates (0x14):            2 Slots, no Reset required
Optional Admin Commands (0x0017):   Security Format Frmw_DL *Other*
Optional NVM Commands (0x001f):     Comp Wr_Unc DS_Mngmt Wr_Zero Sav/Sel_Feat
Maximum Data Transfer Size:         128 Pages
Warning  Comp. Temp. Threshold:     80 Celsius
Critical Comp. Temp. Threshold:     85 Celsius
Namespace 1 Features (0x02):        NA_Fields

Supported Power States
St Op     Max   Active     Idle   RL RT WL WT  Ent_Lat  Ex_Lat
 0 +     5.50W       -        -    0  0  0  0        0       0
 1 +     3.50W       -        -    1  1  1  1        0       0
 2 +     3.00W       -        -    2  2  2  2        0       0
 3 -   0.0700W       -        -    3  3  3  3     4000   10000
 4 -   0.0025W       -        -    4  4  4  4     4000   45000

Supported LBA Sizes (NSID 0x1)`

minwooim, 2019-05-14:
nvme-cli shows a status itself from the device.  Maybe the device does not support get-log command for endurance log.

keithbusch, 2019-05-14:
Right, the tool is just the messenger and only reporting how the device responded to the requested action.

closed 2019-05-14 by keithbusch