linux-nvme/nvme-cli #501: `nvme format --ses=1 /dev/nvme0` destroyed ALL data on ALL SSDs

evilscientress, 2019-05-14:
I used the nvme 1.6 on Gentoo Linux and rand the following commands wich cause a secure erase on BOTH SSDs. 

```
root@somehost /root % nvme format --ses=1 /dev/nvme1
Success formatting namespace:ffffffff
root@somehost /root % nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1             86AS109YTUHV THNSN5512GPU7 TOSHIBA                    1         512.11  GB / 512.11  GB    512   B +  0 B   57GA4103
/dev/nvme1n1     385S1099TYST         KXG50ZNV512G TOSHIBA                     1         512.11  GB / 512.11  GB    512   B +  0 B   AAGA4102
root@somehost /root % nvme format --ses=1 /dev/nvme1n1
Success formatting namespace:1
root@somehost /root % nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1             86AS109YTUHV THNSN5512GPU7 TOSHIBA                    1         512.11  GB / 512.11  GB    512   B +  0 B   57GA4103
/dev/nvme1n1     385S1099TYST         KXG50ZNV512G TOSHIBA                     1         512.11  GB / 512.11  GB    512   B +  0 B   AAGA4102
```

I don't know what cause it but at least there should be a big warning in the documentation that this is possibility. I never expected a tool used on one device to fail in a way that cause data loss on all devices in the system.

keithbusch, 2019-05-14:
Well, it looks like you ran secure erase on two different drives.

keithbusch, 2019-05-14:
Specifically, you may be under the mistaken assumption that /dev/nvme1 has something to do with /dev/nvme1n1. They don't, except by coincidence, so most likely you've run a secure erase format to two completely different devices. That's just how the kernel names their devices when using nvme native multipathing due to subsystem level sharing. If you don't care for this feature, you can disable it with kernel param `nvme_core.multipath=0`.

The tool absolutely will not run commands to a device you didn't ask it to, so your expectations are being met.

evilscientress, 2019-05-14:
So your are telling me that the `/dev/nvme0` and `/dev/nvme1` devices are completely unrelated to `/dev/nvme0n1` and `/dev/nvme1n1`.????

If that is the case:

- why does it read in the manpage: "The <device> parameter is mandatory and may be either the NVMe character device (ex: /dev/nvme0), or a namespace block device (ex: /dev/nvme0n1)." 
- why is there no explicit warning if you specify such a device for a operation that cause irrecoverable. Especially if it's know behavior

keithbusch, 2019-05-15:
The man page does not say nvme0 is the same as nvme0n1. They're just examples of handles you can issue the command to.

You have to be root to issue these commands. With great power comes great responsibility.

keithbusch, 2019-05-15:
In all honesty, I am sorry to hear you've lost data. This happens because the kernel sets up device handle names that are confusing, so you think you may be issuing a command to a namespace's controller, but really a command is going to an unintended device. You are far from the first to hit this quirk, see https://github.com/linux-nvme/nvme-cli/issues/455.

The tool has no part in the device naming. This is entirely on the kernel, and there really is no good way to change this in the without breaking something.

What we can do is have the tool provide additional information in the 'list' command, or maybe other commands, that parse the sysfs hierarchy to group namespaces, controllers and subsystems so that there is an accurate representation of these names and relations.

In the meantime, the sysfs hierarchy is the authority on the names.  Just some examples on how it may be viewed:

```
# ls -l /sys/block/nvme0n1/device
lrwxrwxrwx. 1 root root 0 May 14 15:26 /sys/block/nvme0n1/device -> ../../nvme2

# ls -l /sys/block/nvme2n1/device
lrwxrwxrwx. 1 root root 0 May 14 15:26 /sys/block/nvme2n1/device -> ../../nvme0
```

This is showing that my /dev/nvme0n1 namespace has controller named nvme2, and namespace /dev/nvme2n1 belongs to controller named nvme0. That parent name swizzling is similar to what you've experienced.

This whole naming setup is not intended for pc's, though. This feature is more for servers accessing storage on redundant paths, and we don't find such redundancy in consumer production environments, and so I'd recommend disabling this naming with kernel parameter `nvme_core.multipath=0`.

codingcatgirl, 2019-05-15:
As a software developer, it is, in my opinion, important to inform the user of pitfalls you know of, even if they are not your fault.

Yes, with great power comes great responsibility. I'd say, writing a piece of software that can delete data is great power and comes with the great responsibility as well – the responsibility to take measures to avoid users making easily overlooked mistakes. `rm` will not delete directories without a second parameter. `ddrescue` will not overwrite devices without an extra parameter. Sure, if these tools are not used correctly, you can fuck up. But if you write a tool that you can easily fuck up with, even if the ultimate reason for the easy fuck up is not your fault, it is your ethical responsibility to inform users of possible pitfalls and make it hard to overlook them.

Just my two cents.

keithbusch, 2019-05-15:
If we start requiring additional parameters that weren't there before, we will break factories that build these devices with automation.

codingcatgirl, 2019-05-15:
Right off the top of my head, several options for this:

- Put it into the documentation. That won't break anything.
- Put out a warning if the user is possibly making a mistake and give them a few seconds to read it and possibly cancel it. Inform them that there is an additional parameter that can suppress the message and the delay. This won't break anything. Possibly deprecate usage without the parameter at some point.

I'm sure there are other options. There is a problem here that needs solving and it is causing data loss to users. While this problem is strictly speaking not your responsibility, it sure isn't in your interest that this happens so some kind of solution is desirable. The question is now why not do to it, but how to do it.

evilscientress, 2019-05-15:
As you pointed out, I'm not the first one to who got hit by this. And in the current state I'm definitely not going to be the last one. As @codingcatgirl pointed out adding a warning to the documentation will not break anything. I talked to a lot of experienced Linux users and admins, and not one of them knew about the naming inconsistency. It's nothing anybody would suspect and as you pointed out NVMe multipathing is something quite rare, so users won't even know it's a thing.

For you it's something that is clear and expected behavior, because you are deeply involved with the NVMe subsystem, but for someone who is just using NVMe devices in their server, desktop, laptop this is a total surprise. I did my research and read multiple blog posts and the man page of nvme format before using it. Not in one place was this mentioned. In fact I haven't yet found the documentation for this behavior, and I now know exactly what I'm looking for.

My suggestion would be to at least add a warning in the man pages of destructive commands like format. Because most likely nvme-cli is the first time a user will come in contact with the more in depth parts of the NVMe subsystem.

keithbusch, 2019-05-15:
Documentation is always welcome.

I still think your best option is to disable native nvme multipathing. Non-server distros might ought to do that by default since they're likely not using multipath devices anyway, so have nothing to gain from enabling it.

keithbusch, 2019-05-15:
Let me step back a second to make sure we're talking about the same thing.

I am assuming the only problem is people are not aware `/dev/nvme0` may or may not be the parent controller to namespace `/dev/nvme0n1`.

Is anyone unaware that a low level format command may wipe data from the target?

evilscientress, 2019-05-15:
> I am assuming the only problem is people are not aware /dev/nvme0 may or may not be the parent controller to namespace /dev/nvme0n1.

That's what I meant, sorry if that wasn't clear in my message.

> Is anyone unaware that a low level format command may wipe data from the target?

Nobody I know of. 

And in regards to deactivating NVMe multipathing, at least in my case it would not have made a difference as it was a server. Yes it wasn't hardware that supports multipathing but it was a server OS and Kernel wise. But the strange thing is, while I was rebuilding the server the last couple hours, I checked the kernel and at least my default kernel config has NVMe multipathing disabled and currently the naming of the controller and namespace matches even with multipathing enabled. I don't know if the it's something that changes from boot to boot or is dependent on make and model of the installed SSDs.

keithbusch, 2019-05-15:
As long as multipathing is disabled in the kernel config or param, the namespace handle will align with controller handles.

Mulitpathing assigns namespace device handles as children of subsystems rather than controllers. Subsystem identifiers are generated independently of controller identifiers.

When multipathing is enabled, the name, nvmeXnY, the "X" is the subsystem identifier . The controller handle, /dev/nvmeZ, the "Z" is the controller identifier, independent of the subsystem identifier. If the "Z" and "X" happen to match, it is only by coincidence.

When multipathing is disabled, the namespace parent is the controller, so when examining nvmeXnY, the "X" there comes from controller identifier, and we never use the subsystem identifier.

keithbusch, 2019-05-15:
I'm going to reopen this for now since we don't have a satisfactory resolution.

I think we'll have to attack this from multiple directions. Easiest is through the documentation, and that's no problem to add.

Second, I'd like to augment either 'list' or 'list-subsys' to display device node relationships and multipath'ed namespaces.

Third, I have a kernel patch in progress that tries to create a more sane naming scheme making the char device a subsys handle rather than a controller handle. I'm sure it will break something, but maybe I can break only things no one cares about.

I'm concerned about adding new warnings because such a thing would almost certainly need to include unique identification of the command's destined target device. Those are just cryptic strings (ex: `nqn.2014-08.org.nvmexpress:uuid:f81d4fae-7dec-11d0-a765-00a0c91e6bf6`), and I can't imagine that is going to be helpful to very many people and would invite more confusion.

It may also be best in recommending only sending this command to namespace handles, and have char devices restricted to expert mode, but no real enforcement behind that policy. The admin handle has some pretty wild capabilities to irrevocably modify components that may fall below typical user visibility. Did you know you can destroy namespace data in a subsystem not even attached to the controller you've sent the command to? Yes, that's a spec required feature...

evilscientress, 2019-05-16:
Yeah adding to the documentation seams definitely to be a good thing and low hanging fruit improvement wise. It's also something I could try to help with if help is wanted.

Augmenting `list` or `list-subsys` seams like a good thing too.

Oh a kernel patch, nice. Thanks a lot for putting so much energy into this.

What I meant for my part with adding warnings is something like displaying a warning when a controller device is used instead of a namespace device for a command like format and sleeping for 2-5 sec to give the user a small window to abort. 

From the knowledge I gained in the last 2 days I think that indeed it would be the best to recommend only using namespace handles. Restricting char devices to expert mode sounds like a very good idea to me, especially after reading what else could go wrong in your last comment, and is basically the same thing I meant earlier when I suggested adding an additional flag. The way I thought about it is that the user has to add a `-f` or something similar if they execute a potentially destructive command on a controller instead of a namespace. Restricting char devices to expert mode would achieve the same.

keithbusch, 2019-05-17:
Documentation has been updated.

Kernel patch has been RFC'ed. We can modify names without actually breaking anything and completely avoid mistaken identity issues, though I'm sure someone will still complain: http://lists.infradead.org/pipermail/linux-nvme/2019-May/024142.html

I have a colleague working on updating the 'list' command to show the subsystem topology in greater detail.

We also have support to create udev rules to expose nvme under a new /dev/nvme/ directory.

I am still balking on adding new run-time warnings or chicken-bit options...

evilscientress, 2019-05-18:
awesome, thanks a lot for your work :)

keithbusch, 2019-08-22:
An appropriate warning listing all the relatives of the device being formatted has been committed. There is a significant delay provided to abort the operation. It will continue to format if the user does not take any action to stop it within the provided time.

closed 2019-08-22 by keithbusch

evilscientress, 2019-08-22:
awesome, thanks for your work :)

ak2766, 2023-03-22:
Late to the party, I am.

I read all that and I'm still confused!  Has someone somewhere made a video explaining what this is all about and how not to shoot oneself in the toes?  Or is it now a moot point?  I doubt it's moot as I just came across it in the [CentOS mailing list](https://lists.centos.org/pipermail/centos/2023-March/896737.html).

igaw, 2023-03-22:
Please don't hijack old threads, instead open a new issue.

I quickly tried to figure out what the question is, though the only real info is the link to the mailing list where they are talking about udev rules.

keithbusch, 2023-03-22:
To tack onto @igaw said, the centos issue appears unique to scsi devices, and nvme doesn't share code with that subsystem.

I do think it's moot at this point, though. We have sane nvme handle naming in kernel now aligned to user expectations, so there is substantially lower (zero?) risk of someone formatting the wrong device due to naming confusion.

ak2766, 2023-03-27:
Apologies for the noise - 🙏.