linux-nvme/nvme-cli #502: Build failing on CentOS 6.10

mishagreen, 2019-05-16:
```
[root@box2 nvme-cli]# make
NVME_VERSION = 1.8.1.47.g03df
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1.47.g03df"' -o argconfig.o -c argconfig.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1.47.g03df"' -o suffix.o -c suffix.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1.47.g03df"' -o parser.o -c parser.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1.47.g03df"' -o nvme-print.o -c nvme-print.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1.47.g03df"' -o nvme-ioctl.o -c nvme-ioctl.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1.47.g03df"' -o nvme-lightnvm.o -c nvme-lightnvm.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1.47.g03df"' -o fabrics.o -c fabrics.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1.47.g03df"' -o json.o -c json.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1.47.g03df"' -o nvme-models.o -c nvme-models.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1.47.g03df"' -o plugin.o -c plugin.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1.47.g03df"' -o plugins/intel/intel-nvme.o -c plugins/intel/intel-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1.47.g03df"' -o plugins/lnvm/lnvm-nvme.o -c plugins/lnvm/lnvm-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1.47.g03df"' -o plugins/memblaze/memblaze-nvme.o -c plugins/memblaze/memblaze-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1.47.g03df"' -o plugins/wdc/wdc-nvme.o -c plugins/wdc/wdc-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1.47.g03df"' -o plugins/wdc/wdc-utils.o -c plugins/wdc/wdc-utils.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1.47.g03df"' -o plugins/huawei/huawei-nvme.o -c plugins/huawei/huawei-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1.47.g03df"' -o plugins/netapp/netapp-nvme.o -c plugins/netapp/netapp-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1.47.g03df"' -o plugins/toshiba/toshiba-nvme.o -c plugins/toshiba/toshiba-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1.47.g03df"' -o plugins/micron/micron-nvme.o -c plugins/micron/micron-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1.47.g03df"' -o plugins/seagate/seagate-nvme.o -c plugins/seagate/seagate-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.8.1.47.g03df"' nvme.c -o nvme argconfig.o suffix.o parser.o nvme-print.o nvme-ioctl.o nvme-lightnvm.o fabrics.o json.o nvme-models.o plugin.o plugins/intel/intel-nvme.o plugins/lnvm/lnvm-nvme.o plugins/memblaze/memblaze-nvme.o plugins/wdc/wdc-nvme.o plugins/wdc/wdc-utils.o plugins/huawei/huawei-nvme.o plugins/netapp/netapp-nvme.o plugins/toshiba/toshiba-nvme.o plugins/micron/micron-nvme.o plugins/seagate/seagate-nvme.o
cc1: warnings being treated as errors
nvme.c: In function ‘list_secondary_ctrl’:
nvme.c:2266: error: implicit declaration of function ‘_Static_assert’
make: *** [nvme] Error 1
```

The environment:
```
[root@box2 nvme-cli]# uname -r
2.6.32-754.9.1.el6.x86_64
[root@box2 nvme-cli]# cat /etc/centos-release
CentOS release 6.10 (Final)
[root@box2 nvme-cli]# gcc -v
Using built-in specs.
Target: x86_64-redhat-linux
Configured with: ../configure --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-bootstrap --enable-shared --enable-threads=posix --enable-checking=release --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-languages=c,c++,objc,obj-c++,java,fortran,ada --enable-java-awt=gtk --disable-dssi --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-1.5.0.0/jre --enable-libgcj-multifile --enable-java-maintainer-mode --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --disable-libjava-multilib --with-ppl --with-cloog --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux
Thread model: posix
gcc version 4.4.7 20120313 (Red Hat 4.4.7-23) (GCC)
[root@box2 nvme-cli]# make -v
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for x86_64-redhat-linux-gnu
```

keithbusch, 2019-05-16:
thanks for the notice, will get this fixed today

keithbusch, 2019-05-16:
Fixed now, I missed that macro was included in the pull request. :(

closed 2019-05-16 by keithbusch