linux-nvme/nvme-cli #506: Virtium plugin fails to build

kheitke, 2019-05-23:
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DLIBUUID -DNVME_VERSION='"1.8.1.64.gb969"' -o plugins/virtium/virtium-nvme.o -c plugins/virtium/virtium-nvme.c
plugins/virtium/virtium-nvme.c: In function ‘vt_update_vtview_log_header.isra.2’:
plugins/virtium/virtium-nvme.c:215:108: error: ‘%s’ directive output may be truncated writing up to 16383 bytes into a region of size between 16004 and 16317 [-Werror=format-truncation=]
             "\"devices\":[{\"model\":\"%s\",\"port\":\"%s\",\"SN\":\"%s\",\"type\":\"NVMe\",\"identify\":\"%s\",\"firmwareSlot\":\"%s\"}]}\n", \
                                                                                                            ^~
             header->raw_ctrl.mn, header->path, header->raw_ctrl.sn, identext, fwtext);
                                                                     ~~~~~~~~                                
In file included from /usr/include/stdio.h:867,
                 from plugins/virtium/virtium-nvme.c:3:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:67:10: note: ‘__builtin___snprintf_chk’ output between 91 and 18834 bytes into a destination of size 16384
   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        __bos (__s), __fmt, __va_arg_pack ());
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [Makefile:60: plugins/virtium/virtium-nvme.o] Error 1

minwooim, 2019-05-29:
@kheitke ,

Can this issue be closed?

kheitke, 2019-05-29:
Yes, the issue has been fixed. Thanks

closed 2019-05-29 by kheitke