linux-nvme/nvme-cli #510: nvme naming/association wrong

broth-itk, 2019-06-10:
**Synopsis:**

nvme devices shown in "nvme list" do not match actual devices.
This is a critical issue as it may happen that you accidently format the wrong nvme device!

**How to reproduce:**

Install two nvme ssd drives in a computer (preferably different manufacturers) and compare outputs of "nvme list" with "smartctl -i /dev/nvmeY.
You will see that the names differ.

**Demonstration:**

[root@localhost nvme-cli]# nvme version
nvme version 1.8.1.72.g4fe95
[root@localhost nvme-cli]# cd ..
[root@localhost ~]# nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev  
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     ES85N931210506479    PC401 NVMe SK hynix 256GB                1         256.06  GB / 256.06  GB    512   B +  0 B   80003E00
/dev/nvme1n1     484F7098F1ZP         KXG50ZNV512G NVMe TOSHIBA 512GB          1         512.11  GB / 512.11  GB      4 KiB +  0 B   AADA4106
[root@localhost ~]# nvme list 
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev  
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     ES85N931210506479    PC401 NVMe SK hynix 256GB                1         256.06  GB / 256.06  GB    512   B +  0 B   80003E00
/dev/nvme1n1     484F7098F1ZP         KXG50ZNV512G NVMe TOSHIBA 512GB          1         512.11  GB / 512.11  GB      4 KiB +  0 B   AADA4106
[root@localhost ~]# smartctl -i /dev/nvme0
smartctl 6.5 2016-05-07 r4318 [x86_64-linux-4.14.35-1844.5.3.el7uek.x86_64] (local build)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Number:                       KXG50ZNV512G NVMe TOSHIBA 512GB
Serial Number:                      484F7098F1ZP
Firmware Version:                   AADA4106
PCI Vendor/Subsystem ID:            0x1179
IEEE OUI Identifier:                0x00080d
Total NVM Capacity:                 512,110,190,592 [512 GB]
Unallocated NVM Capacity:           0
Controller ID:                      0
Number of Namespaces:               1
Namespace 1 Size/Capacity:          512,110,190,592 [512 GB]
Namespace 1 Formatted LBA Size:     4096
Local Time is:                      Mon Jun 10 14:43:35 2019 CEST

[root@localhost ~]# smartctl -i /dev/nvme1
smartctl 6.5 2016-05-07 r4318 [x86_64-linux-4.14.35-1844.5.3.el7uek.x86_64] (local build)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Number:                       PC401 NVMe SK hynix 256GB
Serial Number:                      ES85N931210506479
Firmware Version:                   80003E00
PCI Vendor/Subsystem ID:            0x1c5c
IEEE OUI Identifier:                0xace42e
Controller ID:                      1
Number of Namespaces:               1
Namespace 1 Size/Capacity:          256,060,514,304 [256 GB]
Namespace 1 Formatted LBA Size:     512
Local Time is:                      Mon Jun 10 14:44:58 2019 CEST

[root@localhost ~]# uname -a
Linux localhost.localdomain 4.14.35-1844.5.3.el7uek.x86_64 #2 SMP Wed May 8 21:50:52 PDT 2019 x86_64 x86_64 x86_64 GNU/Linux
[root@localhost ~]#

minwooim, 2019-06-10:
On 19-06-10 03:50:52, broth-itk wrote:
> **Synopsis:**
>
> nvme devices shown in "nvme list" do not match actual devices.
> This is a critical issue as it may happen that you accidently format the wrong nvme device!
>
> **How to reproduce:** [...]

I have tried with the exact same version of nvme-cli of yours with
smartmontools v7.0 and it has been reproduced:

root@target:~# nvme --version
nvme version 1.8.1.72.g4fe95
root@target:~# nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     bar                  QEMU NVMe Ctrl                           1         536.87  MB / 536.87  MB    512   B +  0 B   1.0
/dev/nvme1n1     foo                  QEMU NVMe Ctrl                           1         536.87  MB / 536.87  MB    512   B +  0 B   1.0

root@target:~/smartmontools-7.0# ./smartctl -i /dev/nvme0                        
smartctl 7.0 2018-12-30 r4883 [x86_64-linux-5.1.0+] (local build)                
Copyright (C) 2002-18, Bruce Allen, Christian Franke, www.smartmontools.org      
                                                                                 
=== START OF INFORMATION SECTION ===                                             
Model Number:                       QEMU NVMe Ctrl                               
Serial Number:                      foo                                          
Firmware Version:                   1.0                                          
PCI Vendor ID:                      0x8086                                       
PCI Vendor Subsystem ID:            0x1af4                                       
IEEE OUI Identifier:                0xb30200                                     
Controller ID:                      0                                            
Number of Namespaces:               1                                            
Namespace 1 Size/Capacity:          536,870,912 [536 MB]                         
Namespace 1 Formatted LBA Size:     512                                          
Local Time is:                      Mon Jun 10 20:48:44 2019 UTC                 
                                                                                 
root@target:~/smartmontools-7.0# ./smartctl -i /dev/nvme1                        
smartctl 7.0 2018-12-30 r4883 [x86_64-linux-5.1.0+] (local build)                
Copyright (C) 2002-18, Bruce Allen, Christian Franke, www.smartmontools.org      
                                                                                 
=== START OF INFORMATION SECTION ===                                             
Model Number:                       QEMU NVMe Ctrl                               
Serial Number:                      bar                                          
Firmware Version:                   1.0                                          
PCI Vendor ID:                      0x8086                                       
PCI Vendor Subsystem ID:            0x1af4                                       
IEEE OUI Identifier:                0xb30200                                     
Controller ID:                      0                                            
Number of Namespaces:               1                                            
Namespace 1 Size/Capacity:          536,870,912 [536 MB]                         
Namespace 1 Formatted LBA Size:     512                                          
Local Time is:                      Mon Jun 10 20:48:45 2019 UTC                 

Please let me take a closer look here.

broth-itk, 2019-06-10:
Thank you very much for your quick reply and look!
This won't bring my data back but hopefully others will be protected by accidently formatting the wrong NVMe SSD ;)

minwooim, 2019-06-10:
On 19-06-10 03:50:52, broth-itk wrote:
> **Synopsis:**
>
> nvme devices shown in "nvme list" do not match actual devices.
> This is a critical issue as it may happen that you accidently format the wrong nvme device!
>
> **How to reproduce:** [...]

I think it's due to multipath feature in the nvme driver.  If you see
the results from the following commands which is example for three
different controllers:

root@target:~/smartmontools-7.0# nvme list-subsys                                       
nvme-subsys0 - NQN=nqn.2014.08.org.nvmexpress:80861af4bar                 QEMU NVMe Ctrl
\                                                                                       
 +- nvme1 pcie 0000:00:05.0 live                                                        
nvme-subsys1 - NQN=nqn.2014.08.org.nvmexpress:80861af4foo                 QEMU NVMe Ctrl
\                                                                                       
 +- nvme0 pcie 0000:00:04.0 live                                                        
nvme-subsys2 - NQN=nqn.2014.08.org.nvmexpress:80861af4zoo                 QEMU NVMe Ctrl
\                                                                                       
 +- nvme2 pcie 0000:00:06.0 live                                                        

root@target:~/smartmontools-7.0# nvme list                                                                                                   
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev  
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     bar                  QEMU NVMe Ctrl                           1         536.87  MB / 536.87  MB    512   B +  0 B   1.0     
/dev/nvme1n1     foo                  QEMU NVMe Ctrl                           1         536.87  MB / 536.87  MB    512   B +  0 B   1.0     
/dev/nvme2n1     zoo                  QEMU NVMe Ctrl                           1         536.87  MB / 536.87  MB    512   B +  0 B   1.0     

If you do smartctl -i /dev/nvme0n1 instead of /dev/nvme0 which indicates
the nvme subsystem, you can see the results are good to go.

The problem is that the kernel adds a char device with instance number which
is the actual controller id.  But, when it comes to subsystem, (in case
NVME_MULTIPATH enabled) the disk_name is assigned with subsystem id.

I think this issue should be discussed with kernel people that if multipath
is eanbled, mitmatch between controller and namespace device node file should
be matched or not.

minwooim, 2019-06-10:
You can find what Keith has said in the kernel mailing list for nvme:
(Thanks to Keith)
  http://lists.infradead.org/pipermail/linux-nvme/2019-June/024893.html

If you enabled the multipath, you should see the "nvme list-subsys" and
do things based on the subsystem instance id.  Or you just can see the
sysfs for the nvme controller and namespaces.

/sys/class/nvme/nvmeX/nvmeYnZ
	X for the controller instance id
	Y for the subsystem instance id
	Z for the namesapce id

Thanks :)

alex-zadara, 2019-06-23:
@minwooim 
in disconnect_by_device, I see code:
```
device = basename(device);
ret = sscanf(device, "nvme%d", &instance);
...
return remove_ctrl(instance);
```
So this code is incorrect, as we learned from the issue, yes?
because it assumes that X==Y in your comment https://github.com/linux-nvme/nvme-cli/issues/510#issuecomment-500431277?

Can you please confirm?

minwooim, 2019-06-23:
On 19-06-23 01:21:53, Alex Lyakas - Zadara Storage wrote:
> @minwooim
> in disconnect_by_device, I see code:
> ```
> device = basename(device);
> ret = sscanf(device, "nvme%d", &instance);
> ...
> return remove_ctrl(instance);
> ```
> So this code is incorrect, as we [...]

@alex-zadara ,
`disconnect_by_device()` is just getting the instance number from the
chardev which is the subsystem instance if MULTIPATH is enabled.  And
`remove_ctrl()` just deletes the controller with the subsystem instance.

alex-zadara, 2019-06-23:
@minwooim
I am a bit confused. From your comment and also from the http://lists.infradead.org/pipermail/linux-nvme/2019-June/024893.html email thread, I understand that the chardev represents the controller instance, not the subsystem instance. Furthermore, remove_ctrl() does:
```
if (asprintf(&sysfs_path, "/sys/class/nvme/nvme%d/delete_controller",
```
which according to your comment https://github.com/linux-nvme/nvme-cli/issues/510#issuecomment-500431277
```
/sys/class/nvme/nvmeX/nvmeYnZ
X for the controller instance id
Y for the subsystem instance id
Z for the namesapce id
```
is X, i.e., the controller instance id.

Can you please clarify where is the controller instance id and where is the subsystem instance id?

minwooim, 2019-06-24:
@alex-zadara ,
Sorry for making confusions.  IMHO, the X in /dev/nvmeX node are indicating the
controller instance number.

For the sysfs, the X in /sys/class/nvme/nvmeX indicates the controller number
which is exactly the same with the above device node.  In that controller, we just can
see the namespaces formatted either nvmeXnY or nvmeXcYnZ.

nvmeXnY indicates:
- Subsystem instance X with namespace ID Y

nvmeXcYnZ indicates:
- Subsystem instance X with controller ID Y, and namespace ID Z.

We can see the relationships among them by "nvme list-subsys".
```bash
...
nvme-subsys4 - NQN=test2
\
 +- nvme4 loop  live
 +- nvme7 loop  live
```

If we are seeing nvme-subsys4, two different controllers is shown nvme4 and nvme7.
The nvme4 has been registered to the subsystem before nvme7 which means we
can find /dev/nvme4n1 node as a namespace.  By the way, nvme7 has been registered
to the same subsystem whose NQN is exactly the same.  It means that /dev/nvme7n1
will not be able to be found, but we can think that nvme4c7n1 is hidden.
```
root@target:~# ls /dev                                                
autofs           full       loop4               null     nvme6        
block            hpet       loop5               nvme0    nvme7        
bsg              hugepages  loop6               nvme0n1  nvme-fabrics 
cdrom            hwrng      loop7               nvme1    nvram        
char             initctl    loop-control        nvme1n1  port         
console          input      mapper              nvme2    ptmx         
core             kmsg       md0                 nvme2n1  pts          
cpu              log        mem                 nvme3    random       
cpu_dma_latency  loop0      memory_bandwidth    nvme3n1  rfkill       
disk             loop1      mqueue              nvme4    rtc          
dvd              loop2      network_latency     nvme4n1  rtc0         
fd               loop3      network_throughput  nvme5    sda          
```
```
root@target:~# cat /sys/class/nvme/nvme7/nvme4c7n1/hidden
1                                                        
```
Which is MULTIPATH.

alex-zadara, 2019-06-24:
@minwooim 
Thanks for the detailed explanation! I think I understand more or less.

It just seems strange to me that in disconnect_by_device() we do:
- take basename of a device path, like /dev/nvmeYnZ, which will be "nvmeYnZ"
- parse the "Y"
- assume that the controller instance responsible for the device is "nvmeY". In case of multipath, I think it can be another controller as well.

minwooim, 2019-06-24:
@alex-zadara ,
I guess you're right.  If a user gives a namespace node instead of a
controller node, then nvme-cli should do:
- check if the given namespace node exists.
  - If so, this is a not hidden blkdev so that we just can find out the controller node
    to be disconnected.

If it receives a controller node instead of namespace, then it just can directly
disconnect the controller as it is.

jimnordb, 2019-07-14:
I ran a simple nvme format /dev/nvme2 and it formatted my nvme0. Not sure how i will get my data back at this point

keithbusch, 2019-07-15:
No, format on nvme2 does nothing to nvme0. You are confusing controller and namespace handles.

broth-itk, 2019-07-15:
Sorry, thats not true.

I reported the same issue. Formatting nvmeA may in reality format nvmeB.

This is a serious bug to be solved!

-- 
____________________________________
Roth ITK Consulting GmbH
Tumblingerstr. 23
80337 München
Germany

Fon: +49 89 4161450-10
Fax: +49 89 4161450-99
Email: broth@roth-itk.de
Web: http://www.roth-itk.de

Geschäftsführer: Bernhard Roth
HRB 174971, Amtsgericht München
USt-ID: DE261708872

keithbusch, 2019-07-15:
No, you are very much mixing names and making assumptions that are not valid. Please read the format manpage.

keithbusch, 2019-07-15:
See also issue #501

keithbusch, 2019-07-15:
Closing since this is a duplicate of #501. Add future comments to that issue, or even better would be to take this to the Linux nvme kernel mailing list, since that is the component that provides the names to the handles, which has nothing to do with this tool. This program is simply the messenger for what the kernel provides.

closed 2019-07-15 by keithbusch

broth-itk, 2019-07-15:
I don't care about mixing or assumptions.

THERE IS A SERIOUS PROBLEM SOMEWHERE

The risk of data loss is imminent.

There is no question like "do you really want to format nvmeX, namespace
Y, serial number NNNN"

It just wipes your data.

It happend to me serveral times, fortunately with test data only.

RTFM is always a good idea but think about intuitive usage as well.

Thank you!

-- 
____________________________________
Roth ITK Consulting GmbH
Tumblingerstr. 23
80337 München
Germany

Fon: +49 89 4161450-10
Fax: +49 89 4161450-99
Email: broth@roth-itk.de
Web: http://www.roth-itk.de

Geschäftsführer: Bernhard Roth
HRB 174971, Amtsgericht München
USt-ID: DE261708872

broth-itk, 2019-07-15:
Well okay, then please be so kind and add a new feature

"Security question before formatting"

-- 
____________________________________
Roth ITK Consulting GmbH
Tumblingerstr. 23
80337 München
Germany

Fon: +49 89 4161450-10
Fax: +49 89 4161450-99
Email: broth@roth-itk.de
Web: http://www.roth-itk.de

Geschäftsführer: Bernhard Roth
HRB 174971, Amtsgericht München
USt-ID: DE261708872

freezer5, 2019-07-19:
Wait a minute. Why would anyone format an NVMe **controller** (/dev/nvmeX) ?
Format is a namespace command. It makes sense to format a single namespace via /dev/nvmeXnY, or to specify the namespace ID explicitly when sending the format command to a controller.
I believe the problem here is that when specifying a controller name, the default is to format all namespaces (NSID=0xFFFFFFFF).
Formatting all namespaces of a particular controller sounds like a rare edge-case, it should really be explicitly mentioned on the command line ("-n 0xffffffff"). "nvme format /dev/nvme0" without a namespace ID is most probably a user error rather than an intentional format of all namespaces.
Can we just change that default?

broth-itk, 2019-07-19:
Formatting a all namespaces makes sense IMHO as most end-user NVMe
drives have support for one namespace only.

jm2c

-- 
____________________________________
Roth ITK Consulting GmbH
Tumblingerstr. 23
80337 München
Germany

Fon: +49 89 4161450-10
Fax: +49 89 4161450-99
Email: broth@roth-itk.de
Web: http://www.roth-itk.de

Geschäftsführer: Bernhard Roth
HRB 174971, Amtsgericht München
USt-ID: DE261708872

freezer5, 2019-07-19:
@broth-itk , if there's only one namespace then it's all the more reason to specify the namespace block device (/dev/nvmeXnY), that's the one actually displayed in "nvme list".
If you really want to format multiple namespaces on one controller, and you're sure you know what you're doing, use "nvme format /dev/nvmeX -n 0xffffffff".
"nvme format /dev/nvmeX" without a namespace should fail, as odds are it's not what you wanted.

keithbusch, 2019-07-25:
> Wait a minute. Why would anyone format an NVMe **controller** (/dev/nvmeX) ?
> Format is a namespace command. It makes sense to format a single namespace via /dev/nvmeXnY, or to specify the namespace [...]

So this is kind of missing where the confusion is coming from. We do need to allow the format through the controller handle because you *cant* format hidden or unattached namespaces.

The problem is that some people believe that nvme0 is the parent controller to nvme0n1. That simply is not true for any system that has more than one nvme device when using native nvme multipathing.

The parent controller to nvme0n1 may be nvme0, nvme1, nvme2, etc... There simply is no guaranteed relationship based purely on the name of the device handle. You have to query the path either through sysfs or 'nvme list-subsys /dev/nvme0n1' to view the parent controller(s).

freezer5, 2019-07-25:
@keithbusch , agreed, thanks for clarifying that in the latest documentation.
Should 0xFFFFFFFF be the **default**, though?
The NVM Express spec requires the host to specify the namespace ID (or the special value 0xFFFFFFFF) explicitly in the Format command, if you specify an invalid NSID the command is rejected. I believe we should do the same.

keithbusch, 2019-07-25:
> Should 0xFFFFFFFF be the **default**, though?

Hm, I thought we checked ID_CTRL.FNA before defaulting to NSID_ALL. I believe we ought to add that check before defaulting to FFFFFFFFh, and failing without FNA unless namespace id is otherwise specified.

koct9i, 2019-11-10:
It seems device naming for nvme multipath will be changed in linux 5.4
patch "nvme: Assign subsys instance from first ctrl"
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=733e4b69d508d03c20adfdcf4bd27abc60fae9cc

keithbusch, 2019-11-11:
> It seems device naming for nvme multipath will be changed in linux 5.4
> patch "nvme: Assign subsys instance from first ctrl"

Indeed, that commit will make the naming a bit more sane and will definitely prevent mistaken identity. It is still possible to observe controllers with instances that don't match their namespaces, but it is impossible to get a namespace instance that matches a non-owning controller.