linux-nvme/nvme-cli #518: NVME drives have no mechanism for reporting power consumption

rustyscottweber, 2019-07-02:
I'm not sure if this is a result of the NVME drives not providing the number of if it's a result of the nvme driver not getting the watts consumed by the card.  I'm a NAND test engineer and I keep attempting to understand the power usage of many of the components inside of the systems that I build, but am unable to find any metrics reported by any nvme drives about the amount of watts in current use by the card.  This impacts thermal requirements when designing a system and varies wildly from their older spinning disk ancestors.  It's also imperative in many diagnostics cases, like thermal throttling or PCIE brown out, to not only understand the current temperature of the NVME drive, but also understand the amount of power being consumed in watts.  It would make sense that the older smart-ctrl interfaces didn't ever provide this data because the older spinning media would work directly off of specific power states and the power state could be determined by querying the drive, but the power consumed was never enough to cause issues for PCIE and were never really used.  Am I missing something on how to get power usage in a generic way from an NVME drive?  Is this a metric that needs to be added?  How do I go about asking NVME device vendors to kindly start populating this stat?

rustyscottweber, 2019-07-02:
maybe this is the wrong place to post this as a feature request.. In which case I apologize in advance, but I would like to know the correct place if it is not the correct place.

keithbusch, 2019-07-03:
NVMe doesn't provide a generic way to read actual power draw. It only provides max power for a particular power state.

The actual power consumption for any party instant will be platform or devicevendor specific. For example, some platforms can show device power usage through 'powertop'.

rustyscottweber, 2019-07-03:
Almost all of the drives are already recording the amps drawn for thermal throttling, it's only a matter of exposing that sensor to the tool and multiplying it by the current volts. The fusion io drives I used to work on had a utility which would show you current power usage as well as max power draw and it was very useful.  As pcie goes to gen 4.0, it's going to be especially important since the nand's writing ability will only be limited by heat and power. I've also vseen a lot of variance in chassis vendors and their ability to provide meaningful metrics.  Most vendors seem to only be interested in the watts used by the system as a whole.  If you have 12 nvme drives in a system and only one is having a problem drawing power, the total watts being used from the perspective of the psu is going to do you little good. Is there some type of consortium that we can talk to about making this much more standard?

keithbusch, 2019-07-03:
The NVMe Technical Workgroup is the correct forum to standardize new features. Please see nvmexpress.org for information on joining/contributing.

closed 2019-07-03 by keithbusch