linux-nvme/nvme-cli #521: error check missing for ioctl(fd,BLKRRPART) which is actually failing in nvme_format function .

mpacha, 2019-07-18:
#nvme format -n 1 -b 4096  /dev/nvme0
device opened /dev/nvme0
Success formatting namespace:1
IOCTL: Inappropriate ioctl for device

I added perror to see the error message .

minwooim, 2019-07-21:
I think the reason might be due to the character device given.
We should BLKRRPART in case of block device only.  OR we just can
get a new file descriptor for the case for blkdev to BLKRRPART.

Thanks for reporting it.

freezer5, 2019-07-22:
The fix is correct, however there's another underlying problem which is the same as https://github.com/linux-nvme/nvme-cli/issues/510 and https://github.com/linux-nvme/nvme-cli/issues/501 - sending a format command to the character device is problematic. In this case, it might be better to report the namespaces affected by the format and issue BLKRRPART on each attached namespace.

I really think we should block this usage, forcing the user to specify the NSID explicitly (see https://github.com/linux-nvme/nvme-cli/commit/5b0616c0b476e7f63aa255a341d8eacbab4bced0 ).

keithbusch, 2019-07-23:
Will close this, since issue is reported as fixed. Please let's take comments on other issues to the appropriate thread.

closed 2019-07-23 by keithbusch