linux-nvme/nvme-cli #551: nvme-cli: NVMe status: INVALID_FIELD: A reserved coded value or an unsupported value in a defined field(0x4002)

mangowda, 2019-08-16:
Tried running nvme-cli tests on mainline kernel 5.3.0. Test server has 2 NVMe disks (see below for details). Disk IO was tried on the second NVMe disk nvme1n1 before running nvme-cli tests.

[root@ ~]# lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
nvme0n1     259:2    0  5.8T  0 disk 
├─nvme0n1p3 259:5    0   32G  0 part /
├─nvme0n1p1 259:3    0  512M  0 part /boot/efi
├─nvme0n1p4 259:6    0   16G  0 part [SWAP]
└─nvme0n1p2 259:4    0  512M  0 part /boot
nvme1n1     259:0    0  5.8T  0 disk 
└─nvme1n1p1 259:1    0  600G  0 part 
[root@ ~]#

Seeing the below errors:
1. nvme_attach_detach_ns_test.py test failed with the error : 
NVMe status: INVALID_FIELD: A reserved coded value or an unsupported value in a defined field(0x4002)
2. nvme_compare_test.py test failed with the error : 
NVMe status: INVALID_OPCODE: The associated command opcode field is not valid(0x4001)

It would be of great help if some one can provide more info on how to interpret these failures and if there is any issue with NVMe disk config we are using.

keithbusch, 2019-08-16:
The test should be valid for only a subset of existing nvme controllers that support these lesser used opcodes. Perhaps they're not detecting validity before running; will need to verify when I have more time.

mangowda, 2019-08-22:
Thank you Keith for the response. It would be ideal to check validity of the opcodes before running.

closed 2019-09-17 by keithbusch