linux-nvme/nvme-cli #558: nvme cli 1.9 is not getting installed in rhel 8.0

visheshvijay, 2019-08-27:
Hi Keith ,

Terminal prints : ( I have tried both with make clean and make also )

[root@jaguar54 nvme-cli-1.9]# make clean
rm -f nvme argconfig.o suffix.o parser.o nvme-print.o nvme-ioctl.o nvme-lightnvm.o fabrics.o json.o nvme-models.o plugin.o nvme-status.o plugins/intel/intel-nvme.o plugins/lnvm/lnvm-nvme.o plugins/memblaze/memblaze-nvme.o plugins/wdc/wdc-nvme.o plugins/wdc/wdc-utils.o plugins/huawei/huawei-nvme.o plugins/netapp/netapp-nvme.o plugins/toshiba/toshiba-nvme.o plugins/micron/micron-nvme.o plugins/seagate/seagate-nvme.o plugins/virtium/virtium-nvme.o plugins/shannon/shannon-nvme.o *~ a.out NVME-VERSION-FILE *.tar* nvme.spec version control nvme-*.deb
make -C Documentation clean
make[1]: Entering directory '/root/nvme-cli-1.9/Documentation'
rm -f *.xml *.xml+ *.html+
rm -f *.texi *.texi+ *.texi++ nvme.info nvmeman.info
rm -f *.pdf
rm -f howto-index.txt howto/*.html doc.dep
rm -f technical/*.html technical/api-index.txt
rm -f cmds-ancillaryinterrogators.txt cmds-ancillarymanipulators.txt cmds-mainporcelain.txt cmds-plumbinginterrogators.txt cmds-plumbingmanipulators.txt cmds-synchingrepositories.txt cmds-synchelpers.txt cmds-purehelpers.txt cmds-foreignscminterface.txt *.made
rm -f manpage-base-url.xsl
rm -f *~
make[1]: Leaving directory '/root/nvme-cli-1.9/Documentation'
rm -f tests/*.pyc
rm -f verify-no-dep

[root@jaguar54 nvme-cli-1.9]# make install
NVME_VERSION = 1.9
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o argconfig.o -c argconfig.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o suffix.o -c suffix.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o parser.o -c parser.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o nvme-print.o -c nvme-print.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o nvme-ioctl.o -c nvme-ioctl.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o nvme-lightnvm.o -c nvme-lightnvm.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o fabrics.o -c fabrics.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o json.o -c json.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o nvme-models.o -c nvme-models.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugin.o -c plugin.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o nvme-status.o -c nvme-status.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/intel/intel-nvme.o -c plugins/intel/intel-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/lnvm/lnvm-nvme.o -c plugins/lnvm/lnvm-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/memblaze/memblaze-nvme.o -c plugins/memblaze/memblaze-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/wdc/wdc-nvme.o -c plugins/wdc/wdc-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/wdc/wdc-utils.o -c plugins/wdc/wdc-utils.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/huawei/huawei-nvme.o -c plugins/huawei/huawei-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/netapp/netapp-nvme.o -c plugins/netapp/netapp-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/toshiba/toshiba-nvme.o -c plugins/toshiba/toshiba-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/micron/micron-nvme.o -c plugins/micron/micron-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/seagate/seagate-nvme.o -c plugins/seagate/seagate-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/virtium/virtium-nvme.o -c plugins/virtium/virtium-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/shannon/shannon-nvme.o -c plugins/shannon/shannon-nvme.c
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' nvme.c -o nvme argconfig.o suffix.o parser.o nvme-print.o nvme-ioctl.o nvme-lightnvm.o fabrics.o json.o nvme-models.o plugin.o nvme-status.o plugins/intel/intel-nvme.o plugins/lnvm/lnvm-nvme.o plugins/memblaze/memblaze-nvme.o plugins/wdc/wdc-nvme.o plugins/wdc/wdc-utils.o plugins/huawei/huawei-nvme.o plugins/netapp/netapp-nvme.o plugins/toshiba/toshiba-nvme.o plugins/micron/micron-nvme.o plugins/seagate/seagate-nvme.o plugins/virtium/virtium-nvme.o plugins/shannon/shannon-nvme.o
install -d /usr/sbin
install -m 755 nvme /usr/sbin
make -C Documentation install-no-build
make[1]: Entering directory '/root/nvme-cli-1.9/Documentation'
install -d -m 755 /usr/local/share/man/man1
install -m 644 nvme-wdc-vs-internal-log.1 nvme-wdc-purge-monitor.1 nvme-virtium-show-identify.1 nvme-get-log.1 nvme-admin-passthru.1 nvme-disconnect-all.1 nvme-get-feature.1 nvme-dir-send.1 nvme-delete-ns.1 nvme-lnvm-info.1 nvme-ns-rescan.1 nvme-read.1 nvme-wdc-drive-essentials.1 nvme-lnvm-remove.1 nvme-wdc-purge.1 nvme-subsystem-reset.1 nvme-list-subsys.1 nvme-resv-register.1 nvme-changed-ns-list-log.1 nvme-intel-internal-log.1 nvme-reset.1 nvme-disconnect.1 nvme-write.1 nvme-resv-acquire.1 nvme-huawei-list.1 nvme-set-property.1 nvme-io-passthru.1 nvme-dir-receive.1 nvme-fw-log.1 nvme-ana-log.1 nvme-device-self-test.1 nvme-id-ctrl.1 nvme-id-ns.1 nvme-wdc-get-drive-status.1 nvme-wdc-vs-smart-add-log.1 nvme-show-regs.1 nvme-netapp-smdevices.1 nvme-security-recv.1 nvme-lnvm-init.1 nvme-connect-all.1 nvme-connect.1 nvme-lnvm-factory.1 nvme-compare.1 nvme-list.1 nvme-intel-smart-log-add.1 nvme-help.1 nvme-toshiba-clear-pcie-correctable-errors.1 nvme-smart-log.1 nvme-create-ns.1 nvme-wdc-get-crash-dump.1 nvme-security-send.1 nvme-get-property.1 nvme-intel-temp-stats.1 nvme-huawei-id-ctrl.1 nvme-attach-ns.1 nvme-error-log.1 nvme-wdc-vs-nand-stats.1 nvme-id-nvmset.1 nvme-sanitize.1 nvme-wdc-drive-log.1 nvme-sanitize-log.1 nvme-wdc-clear-assert-dump.1 nvme-telemetry-log.1 nvme-lnvm-create.1 nvme-detach-ns.1 nvme-netapp-ontapdevices.1 nvme-lnvm-list.1 nvme-fw-commit.1 nvme-dsm.1 nvme-intel-market-name.1 nvme-effects-log.1 nvme-list-ns.1 nvme-lnvm-diag-set-bbtbl.1 nvme-gen-hostnqn.1 nvme-list-ctrl.1 nvme-toshiba-vs-smart-add-log.1 nvme-format.1 nvme-lnvm-id-ns.1 nvme-lnvm-diag-bbtbl.1 nvme-wdc-drive-resize.1 nvme-discover.1 nvme-write-uncor.1 nvme-wdc-cap-diag.1 nvme-get-ns-id.1 nvme-resv-report.1 nvme-toshiba-vs-internal-log.1 nvme-set-feature.1 nvme-self-test-log.1 nvme-wdc-clear-pcie-correctable-errors.1 nvme-endurance-log.1 nvme-resv-release.1 nvme-fw-download.1 nvme-intel-lat-stats.1 nvme-wdc-get-pfail-dump.1 nvme-virtium-save-smart-to-vtview-log.1 nvme-write-zeroes.1 nvme-flush.1 nvme-wdc-id-ctrl.1 nvme-ns-descs.1 nvme-intel-id-ctrl.1 nvme.1 /usr/local/share/man/man1
install: cannot stat 'nvme-wdc-vs-internal-log.1': No such file or directory
make[1]: *** [Makefile:118: install-no-build] Error 1
make[1]: Leaving directory '/root/nvme-cli-1.9/Documentation'
make: *** [Makefile:86: install-man] Error 2
[root@jaguar54 nvme-cli-1.9]#

keithbusch, 2019-08-28:
tests fine on our end from the git repo. I downloaded the "release" tgz file, that also seems fine.

visheshvijay, 2019-08-29:
Hi Keith,

I have tried it again.
This time I extracted the zip file instead of tar and tried installation Again I observed issue.

Can you please have a look :

> [root@jaguar54 nvme-cli-1.9]# make
> NVME_VERSION = 1.9
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o argconfig.o -c argconfig.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o suffix.o -c suffix.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o parser.o -c parser.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o nvme-print.o -c nvme-print.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o nvme-ioctl.o -c nvme-ioctl.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o nvme-lightnvm.o -c nvme-lightnvm.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o fabrics.o -c fabrics.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o json.o -c json.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o nvme-models.o -c nvme-models.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugin.o -c plugin.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o nvme-status.o -c nvme-status.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/intel/intel-nvme.o -c plugins/intel/intel-nvme.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/lnvm/lnvm-nvme.o -c plugins/lnvm/lnvm-nvme.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/memblaze/memblaze-nvme.o -c plugins/memblaze/memblaze-nvme.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/wdc/wdc-nvme.o -c plugins/wdc/wdc-nvme.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/wdc/wdc-utils.o -c plugins/wdc/wdc-utils.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/huawei/huawei-nvme.o -c plugins/huawei/huawei-nvme.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/netapp/netapp-nvme.o -c plugins/netapp/netapp-nvme.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/toshiba/toshiba-nvme.o -c plugins/toshiba/toshiba-nvme.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/micron/micron-nvme.o -c plugins/micron/micron-nvme.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/seagate/seagate-nvme.o -c plugins/seagate/seagate-nvme.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/virtium/virtium-nvme.o -c plugins/virtium/virtium-nvme.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' -o plugins/shannon/shannon-nvme.o -c plugins/shannon/shannon-nvme.c
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DNVME_VERSION='"1.9"' nvme.c -o nvme argconfig.o suffix.o parser.o nvme-print.o nvme-ioctl.o nvme-lightnvm.o fabrics.o json.o nvme-models.o plugin.o nvme-status.o plugins/intel/intel-nvme.o plugins/lnvm/lnvm-nvme.o plugins/memblaze/memblaze-nvme.o plugins/wdc/wdc-nvme.o plugins/wdc/wdc-utils.o plugins/huawei/huawei-nvme.o plugins/netapp/netapp-nvme.o plugins/toshiba/toshiba-nvme.o plugins/micron/micron-nvme.o plugins/seagate/seagate-nvme.o plugins/virtium/virtium-nvme.o plugins/shannon/shannon-nvme.o
> [root@jaguar54 nvme-cli-1.9]# make install
> install -d /usr/sbin
> install -m 755 nvme /usr/sbin
> make -C Documentation install-no-build
> make[1]: Entering directory '/root/nvme-cli-1.9/Documentation'
> install -d -m 755 /usr/local/share/man/man1
> install -m 644 nvme-wdc-vs-internal-log.1 nvme-wdc-purge-monitor.1 nvme-virtium-show-identify.1 nvme-get-log.1 nvme-admin-passthru.1 nvme-disconnect-all.1 nvme-get-feature.1 nvme-dir-send.1 nvme-delete-ns.1 nvme-lnvm-info.1 nvme-ns-rescan.1 nvme-read.1 nvme-wdc-drive-essentials.1 nvme-lnvm-remove.1 nvme-wdc-purge.1 nvme-subsystem-reset.1 nvme-list-subsys.1 nvme-resv-register.1 nvme-changed-ns-list-log.1 nvme-intel-internal-log.1 nvme-reset.1 nvme-disconnect.1 nvme-write.1 nvme-resv-acquire.1 nvme-huawei-list.1 nvme-set-property.1 nvme-io-passthru.1 nvme-dir-receive.1 nvme-fw-log.1 nvme-ana-log.1 nvme-device-self-test.1 nvme-id-ctrl.1 nvme-id-ns.1 nvme-wdc-get-drive-status.1 nvme-wdc-vs-smart-add-log.1 nvme-show-regs.1 nvme-netapp-smdevices.1 nvme-security-recv.1 nvme-lnvm-init.1 nvme-connect-all.1 nvme-connect.1 nvme-lnvm-factory.1 nvme-compare.1 nvme-list.1 nvme-intel-smart-log-add.1 nvme-help.1 nvme-toshiba-clear-pcie-correctable-errors.1 nvme-smart-log.1 nvme-create-ns.1 nvme-wdc-get-crash-dump.1 nvme-security-send.1 nvme-get-property.1 nvme-intel-temp-stats.1 nvme-huawei-id-ctrl.1 nvme-attach-ns.1 nvme-error-log.1 nv [... cut ...]
> install -d /usr/lib/systemd/system
> install -m 644 ./nvmf-autoconnect/systemd/* /usr/lib/systemd/system
> install -d /usr/lib/udev/rules.d
> install -m 644 ./nvmf-autoconnect/udev-rules/* /usr/lib/udev/rules.d
> install -d /usr/lib/dracut/dracut.conf.d
> install -m 644 ./nvmf-autoconnect/dracut-conf/* /usr/lib/dracut/dracut.conf.d
> if [ ! -s /etc/nvme/hostnqn ]; then \
>         echo `/usr/sbin/nvme gen-hostnqn` > /etc/nvme/hostnqn; \
> fi
> if [ ! -s /etc/nvme/hostid ]; then \
>         uuidgen > /etc/nvme/hostid; \
> fi
>

keithbusch, 2019-08-29:
O_o
That doesn't look anything like the previous output. In fact, that looks like a successful install.

visheshvijay, 2019-09-03:
Hi Keith ,

Even after successfull install my cli is not updated

[root@jaguar54 ~]# nvme --version
nvme version 1.7

Thanks,
Vishesh

keithbusch, 2019-09-03:
Sounds like your 1.7 is installed in a path of higher precedence than 1.9. Find it by running '`which nvme`', delete it, then update your paths with '`hash`'.

visheshvijay, 2019-09-04:
Thanks it worked now.

closed 2019-09-04 by visheshvijay