linux-nvme/nvme-cli #560: make fails on missing linux/uuid.h

kscotts, 2019-08-30:
line 20 of linux/nvme.h includes <linux/uuid.h> which is missing from the linux/ dir. Commenting this line allows make to finish without error.

yuanyang1234, 2019-09-02:
I met the similar issue. CentOS release 6.5 (Final) + nvme-cli-1.9

> [root@localhost nvme-cli]# make
> cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -DLIBUUID -DNVME_VERSION='"1.4.dirty"' -c nvme.c
> In file included from nvme-print.h:4,
>                  from nvme.c:48:
> nvme.h:27:23: error: uuid/uuid.h: No such file or directory
> In file included from nvme.h:34,
>                  from nvme-print.h:4,
>                  from nvme.c:48:
> linux/nvme.h:19:24: error: linux/uuid.h: No such file or directory
> In file included from nvme.h:34,
>                  from nvme-print.h:4,
>                  from nvme.c:48:
> linux/nvme.h:960: error: expected specifier-qualifier-list before ‘uuid_t’
> nvme.c: In function ‘gen_hostnqn_cmd’:
> nvme.c:3236: error: ‘uuid_t’ undeclared (first use in this function)
> nvme.c:3236: error: (Each undeclared identifier is reported only once
> nvme.c:3236: error: for each function it appears in.)
> nvme.c:3236: error: expected ‘;’ before ‘uuid’
> cc1: warnings being treated as errors
> nvme.c:3239: error: implicit declaration of function ‘uuid_generate_random’
> nvme.c:3239: error: ‘uuid’ undeclared (first use in this function)
> nvme.c:3240: error: implicit declaration of function ‘uuid_unparse_lower’
> make: *** [nvme.o] Error 1

keithbusch, 2019-09-03:
I think libuuid is the package for this, not uuid or uuid-devel (at least that's the case on fedora).

kscotts, 2019-09-03:
Also resolves when I use the direct path on sles11sp4 and sles12sp2 "/usr/include/uuid/uuid.h" as shown below. I had to remove the endif: 
/* #include <linux/uuid.h> */
#include </usr/include/uuid/uuid.h>

locate uuid.h
/usr/include/uuid/uuid.h

 # uname -r
3.0.101-108.87-default

# rpm -qif /usr/include/uuid/uuid.h
Name        : libuuid-devel                Relocations: (not relocatable)
Version     : 2.19.1                            Vendor: SUSE LINUX Products GmbH, Nuernberg, Germany
Release     : 6.72.1                        Build Date: Tue Jun  9 00:52:40 2015
Install Date: Wed Dec 13 11:49:11 2017         Build Host: sheep03
Group       : System/Base                   Source RPM: util-linux-2.19.1-6.72.1.src.rpm
Size        : 17085                            License: BSD 3-Clause; GPL v2 or later
Signature   : RSA/8, Tue Jun  9 00:52:59 2015, Key ID e3a5c360307e3d54
Packager    : http://bugs.opensuse.org
URL         : http://userweb.kernel.org/~kzak/util-linux-ng/
Summary     : A collection of basic system utilities
Description :
This package contains a large variety of low-level system utilities
that are necessary for a Linux system to function. It contains the
mount program, the fdisk configuration tool, and more.

Authors:
--------
    Karel Zak <kzak@redhat.com>
Distribution: SUSE Linux Enterprise 11

#############################
# SLES 12
#############################
 uname -r
4.4.121-92.117-default

> rpm -qif /usr/include/uuid/uuid.h
Name        : libuuid-devel
Version     : 2.28
Release     : 44.14.2
Architecture: x86_64
Install Date: Mon 20 Aug 2018 10:25:41 AM PDT
Group       : Development/Libraries/C and C++
Size        : 17123
License     : GPL-2.0+
Signature   : RSA/SHA256, Fri 01 Dec 2017 04:19:39 AM PST, Key ID 70af9e8139db7c82
Source RPM  : util-linux-2.28-44.14.2.src.rpm
Build Date  : Fri 01 Dec 2017 04:18:39 AM PST
Build Host  : sheep09
Relocations : (not relocatable)
Packager    : https://www.suse.com/
Vendor      : SUSE LLC <https://www.suse.com/>
URL         : https://www.kernel.org/pub/linux/utils/util-linux/
Summary     : Development files for libuuid1
Description :
Files to develop applications using the library to generate universally
unique IDs (UUIDs).
Distribution: SUSE Linux Enterprise 12

###############
#### FAILURE ####
###############
> make
NVME_VERSION = 1.9.33.gccfd.dirty
cc -D_GNU_SOURCE -D__CHECK_ENDIAN__ -O2 -g -Wall -Werror -std=gnu99 -I. -DLIBUUID -DNVME_VERSION='"1.9.33.gccfd.dirty"' -Iutil -o nvme-print.o -c nvme-print.c
In file included from nvme.h:27:0,
                 from nvme-print.h:4,
                 from nvme-print.c:7:
linux/nvme.h:1304:2: error: unknown type name ‘uuid_t’
  uuid_t  hostid;
  ^
nvme-print.c:113:40: error: unknown type name ‘uuid_t’
 static const char *nvme_uuid_to_string(uuid_t uuid)
                                        ^
nvme-print.c: In function ‘json_nvme_id_ns_descs’:
nvme-print.c:888:3: error: unknown type name ‘uuid_t’
   uuid_t uuid;
   ^
nvme-print.c:929:4: error: passing argument 1 of ‘memcpy’ makes pointer from integer without a cast [-Werror]
    memcpy(desc.uuid, data + off, sizeof(desc.uuid));
    ^
In file included from nvme-print.c:3:0:
/usr/include/string.h:42:14: note: expected ‘void * restrict’ but argument is of type ‘int’
 extern void *memcpy (void *__restrict __dest, const void *__restrict __src,
              ^
nvme-print.c:930:4: error: implicit declaration of function ‘uuid_unparse_lower’ [-Werror=implicit-function-declaration]
    uuid_unparse_lower(desc.uuid, json_str);
    ^
nvme-print.c: In function ‘show_nvme_id_ns_descs’:
nvme-print.c:975:2: error: unknown type name ‘uuid_t’
  uuid_t uuid;
  ^
nvme-print.c:1006:4: error: passing argument 1 of ‘memcpy’ makes pointer from integer without a cast [-Werror]
    memcpy(uuid, data + pos + sizeof(*cur), 16);
    ^
In file included from nvme-print.c:3:0:
/usr/include/string.h:42:14: note: expected ‘void * restrict’ but argument is of type ‘int’
 extern void *memcpy (void *__restrict __dest, const void *__restrict __src,
              ^
nvme-print.c: In function ‘show_nvme_id_uuid_list’:
nvme-print.c:1396:3: error: unknown type name ‘uuid_t’
   uuid_t uuid;
   ^
nvme-print.c:1422:3: error: implicit declaration of function ‘nvme_uuid_to_string’ [-Werror=implicit-function-declaration]
   printf("UUID         : %s", nvme_uuid_to_string(uuid));
   ^
nvme-print.c:1422:3: error: format ‘%s’ expects argument of type ‘char *’, but argument 2 has type ‘int’ [-Werror=format=]
nvme-print.c: In function ‘json_nvme_id_uuid_list’:
nvme-print.c:1438:3: error: unknown type name ‘uuid_t’
   uuid_t uuid;
   ^
cc1: all warnings being treated as errors
Makefile:70: recipe for target 'nvme-print.o' failed
make: *** [nvme-print.o] Error 1

birkelund, 2019-09-04:
Looks like commit 70501d661cdb is the issue. `linux/uuid.h` does not provide `uuid_t`, but `uuid/uuid.h` does. Reverting that commit fixes the issue for me.

tytso, 2019-09-04:
I have this problem when building on Debian --- linux/uuid.h does *not* provide uuid_t.   I don't know why the commit description claims that using uuid/uuid.h was a "typo"; it works just fine, although it does require that the uuid-dev package be installed.    In any case, linux/uuid.h is **not** a reliable provider for uuid_t.

keithbusch, 2019-09-04:
Yeah, my bad. I'll sort this out.

keithbusch, 2019-09-04:
Sorry all, I made a real mess of this. I've reverted the broken churn back to where we were before this started, then relocated the uuid typedef to a proper place if uuid is not available. This should compile whether uuid-dev package is installed or not.

minwooim, 2019-09-05:
Thanks @keithbusch !