linux-nvme/nvme-cli #563: nvme disconnect -d fails to disconnect nvme block device

pszczerx, 2019-09-04:
### Issue description
The latest version of nvme CLI tool fails to disconnect nvme block device.
Example:
`$ nvme disconnect -d /dev/nvme0n1`
The command will succeed only if you pass NVMe character device (controller) as an argument.
Example:
`$ sudo nvme disconnect -d /dev/nvme0`

As far as I can tell, this issue was introduced by commit: 986a9357
`strcmp()` function in line https://github.com/linux-nvme/nvme-cli/commit/986a9357e27dbbde4b5dbdbe8bb490130465d198#diff-43e749b3fb33e89f12c38df0ad81a1c5R203 returns a non-zero return value, because content of `char d[64];`  array does not contain namespace suffix (i.e.` device=nvme0n1 d=nvme0`).

Older version of nvme CLI tool used to disconnect the controller when you passed a path to nvme block device.

### nvme CLI help snippet
```
nvme --help                 
nvme-1.9.9.g5221                                                   
usage: nvme <command> [<device>] [<args>]                     
                                                              
The '<device>' may be either an NVMe character device (ex: /dev/nvme0) or an
nvme block device (ex: /dev/nvme0n1). 
```
  
### Version Information
```
$ sudo nvme version
nvme version 1.9.9.g5221
```
### Steps to reproduce

```
$ sudo nvme list | cut -d ' ' -f 1
Node
----------------
/dev/nvme0n1
/dev/nvme0n2
/dev/nvme0n3
/dev/nvme0n4
/dev/nvme0n5
/dev/nvme0n6
/dev/nvme0n7

$ sudo nvme disconnect -d /dev/nvme0n1
Failed to disconnect by device name: /dev/nvme0n1
$ 
$ sudo nvme list | cut -d ' ' -f 1
Node
----------------
/dev/nvme0n1
/dev/nvme0n2
/dev/nvme0n3
/dev/nvme0n4
/dev/nvme0n5
/dev/nvme0n6
/dev/nvme0n7

$ sudo nvme disconnect -d /dev/nvme0q
$ sudo nvme list | cut -d ' ' -f 1
$
```

### Logs from nvme CLI 1.7
```
$ sudo nvme version
nvme version 1.7
$ sudo nvme list | cut -d ' ' -f 1
Node
----------------
/dev/nvme0n1
/dev/nvme0n2
/dev/nvme0n3
/dev/nvme0n4
/dev/nvme0n5
/dev/nvme0n6
/dev/nvme0n7
$ sudo nvme disconnect -d /dev/nvme0n1
$ sudo nvme list
$ 
```

Could you clarify whether this change was intentional?

Thanks for help,
Przemek

keithbusch, 2019-09-04:
I think that was intentional due to the namespace and character name swizzling. I'll double check with the author.

keithbusch, 2019-09-04:
Okay, the removal is done by writing to /sys/class/nvme/nvme<X>/delete_controler, but you can't get to that unique path from a namespace handle with multipathing because the namespace may have more than one controller. We wouldn't be able to figure out the intended one to remove from a namespace handle.

pszczerx, 2019-09-10:
Thanks for looking into the issue @keithbusch. If this change was intentional, then I think this issue can be closed.

closed 2019-09-10 by pszczerx