linux-nvme/nvme-cli #567: nvme json output temperature unit

cyprix, 2019-09-09:
In the main smart-log temperature unit in Celcius:
```
nvme smart-log /dev/nvme0n1 | grep -i temper
 temperature                         : 28 C
 Warning Temperature Time            : 0
 Critical Composite Temperature Time : 0
 Temperature Sensor 1                : 38 C
```

But the same command with --output-format=json show unit in Kelvin.
```
nvme smart-log --output-format=json /dev/nvme0n1 | grep -i temp
  "temperature" : 301,
  "warning_temp_time" : 0,
  "temperature_sensor_1" : 312,
```

In the **nvme-print.c** file, I see that you convert the unit from Kelvin to Celcius, in the standard output:
```
	unsigned int temperature = ((smart->temperature[1] << 8) |
		smart->temperature[0]) - 273;
```

Please use in the **--output-format=json** Celcius instead of Kelvin.
[https://github.com/linux-nvme/nvme-cli/blob/master/nvme-print.c#L3191
](https://github.com/linux-nvme/nvme-cli/blob/master/nvme-print.c#L3191
)

I'm not so familiar with C. I changed code line and JSON output format show Celcius, but Temperature sensor still shows the Kelvin.

cyprix, 2019-09-09:
https://github.com/linux-nvme/nvme-cli/pull/568

p-e-w, 2019-12-07:
Changing this now sounds like a really bad idea. Currently, programs parsing the JSON output need to subtract 273 from the temperature value in order to get the temperature in degrees Celsius. This is certainly weird, and probably not how it should be ideally.

But with this proposed change, programs consuming the JSON in the future will need to *check the version of nvme-cli* in order to interpret the output correctly. This would be much, much worse than the status quo. The train for simply switching units left a long time ago. Since the purpose of the JSON output is to be consumed by other programs, interface stability beats convenience here.

keithbusch, 2019-12-07:
If we add an option, like verbose or human-readable or something, then we can use that to print parameters differently

rx80, 2021-09-25:
How about adding a "temperature_unit" entry?

closed 2022-01-19 by igaw