linux-nvme/nvme-cli #582: Running nvme list can lead to segmentation fault

pszczerx, 2019-09-27:
### Issue description

`nvme list` can in certain cases lead to segmentation fault. The issue is reproducible when nvme subsystem contains a symbolic link to non-existing directory. nvme-cli attempts to `scandir()` `/sys/class/nvme-subsystem/nvme-subsys0/nvme0`. Function call fails, but return value is not checked and we end up calling `free()` on uninitialized `ns` pointer which causes segmentation fault in line https://github.com/linux-nvme/nvme-cli/blob/master/nvme-topology.c#L164.

### Version Information

```
$ nvme version
nvme version 1.9.72.g5c1a
```

### Logs 

```
$ sudo nvme list
Failed to open /sys/class/nvme-subsystem/nvme-subsys0/nvme0/address: No such file or directory
Failed to open /sys/class/nvme-subsystem/nvme-subsys0/nvme0/transport: No such file or directory
Failed to open /sys/class/nvme-subsystem/nvme-subsys0/nvme0/state: No such file or directory
Segmentation fault
```
```
Core was generated by `nvme list'.
Program terminated with signal 11, Segmentation fault.
#0  __GI___libc_free (mem=0x6) at malloc.c:2941
2941      if (chunk_is_mmapped(p))                       /* release mmapped memory. */
(gdb) bt
#0  __GI___libc_free (mem=0x6) at malloc.c:2941
#1  0x0000000000427e18 in scan_ctrl (c=0x105b250, p=0x105b0e0 "/sys/class/nvme-subsystem/nvme-subsys0") at nvme-topology.c:164
#2  0x0000000000427fb7 in scan_subsystem (s=0x105afe0) at nvme-topology.c:204
#3  0x000000000042864f in scan_subsystems (t=0x7ffcfce3bf80) at nvme-topology.c:342
#4  0x0000000000405471 in list (argc=1, argv=0x7ffcfce3c240, cmd=0x66a300 <list_cmd>, plugin=0x66ad60 <builtin>) at nvme.c:1407
#5  0x00000000004271e2 in handle_plugin (argc=1, argv=0x7ffcfce3c240, plugin=0x66ad60 <builtin>) at plugin.c:158
#6  0x000000000040f460 in main (argc=2, argv=0x7ffcfce3c238) at nvme.c:5078
(gdb) frame 1
#1  0x0000000000427e18 in scan_ctrl (c=0x105b250, p=0x105b0e0 "/sys/class/nvme-subsystem/nvme-subsys0") at nvme-topology.c:164
164             free(ns);
(gdb) info locals
n = 0x401eca <_start>
ns = 0x6
path = 0x105c330 "/sys/class/nvme-subsystem/nvme-subsys0/nvme0"
i = -1
fd = 32706
ret = 44

```

pszczerx, 2019-09-27:
I'll submit soon a fix for this issue.

closed 2019-10-07 by keithbusch

moshe-y, 2021-06-28:
issue appears again on  5.8.0-55-generic #62~20.04.1-Ubuntu

PR #583 (2019-10-07): Check return value of scandir
On error, scandir returns -1 and does not allocate memory
for namelist array. In some places in the code return value
of scandir call is not checked. This causes nvme-cli to
attempt to free() an uninitialized pointer, which subsequently
leads to segmentation fault.

To address this issue, check return value of scandir calls
throughout the code.

Fixes #582