linux-nvme/nvme-cli #590: nvme cli 1.9.88.g6275: Unable to discover more than 3 volumes

tabasumtamboli, 2019-10-10:
I have connected 4 volumes to a controller, but nvme discovery shows only 3 volumes and rest all entries are garbage
[root@host]# ./nvme version
nvme version 1.9.88.g6275
[root@host]# ./nvme discover -t rdma -a 192.168.6.10

Discovery Log Number of Records 4, Generation counter 74
=====Discovery Log Entry 0======
trtype:  rdma
adrfam:  ipv4
subtype: nvme subsystem
treq:    not specified
portid:  164
trsvcid: 4420
subnqn:  GB00002604bbf973
traddr:  192.168.6.10
rdma_prtype: unrecognized
rdma_qptype: unrecognized
rdma_cms:    unrecognized
rdma_pkey: 0x0000
=====Discovery Log Entry 1======
trtype:  rdma
adrfam:  ipv4
subtype: nvme subsystem
treq:    not specified
portid:  164
trsvcid: 4420
subnqn:  GB00002604bbf974
traddr:  192.168.6.10
rdma_prtype: unrecognized
rdma_qptype: unrecognized
rdma_cms:    unrecognized
rdma_pkey: 0x0000
=====Discovery Log Entry 2======
trtype:  rdma
adrfam:  ipv4
subtype: nvme subsystem
treq:    not specified
portid:  164
trsvcid: 4420
subnqn:  GB00002604bbf975
traddr:  192.168.6.10
rdma_prtype: unrecognized
rdma_qptype: unrecognized
rdma_cms:    unrecognized
rdma_pkey: 0x0000
**=====Discovery Log Entry 3======
trtype:  unrecognized
adrfam:  pci
subtype: unrecognized
treq:    not specified
portid:  0
trsvcid:    <============  Garbage
subnqn:   <============
traddr:  <============**
-----------------

[root@host]# uname -a
Linux host 3.10.0-957.el7.x86_64 #1 SMP Thu Nov 8 23:39:32 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
-----------------
[root@host]# cat /etc/*release
CentOS Linux release 7.6.1810 (Core)
NAME="CentOS Linux"
VERSION="7 (Core)"
ID="centos"
ID_LIKE="rhel fedora"
VERSION_ID="7"
PRETTY_NAME="CentOS Linux 7 (Core)"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:centos:centos:7"
HOME_URL="https://www.centos.org/"
BUG_REPORT_URL="https://bugs.centos.org/"

CENTOS_MANTISBT_PROJECT="CentOS-7"
CENTOS_MANTISBT_PROJECT_VERSION="7"
REDHAT_SUPPORT_PRODUCT="centos"
REDHAT_SUPPORT_PRODUCT_VERSION="7"

CentOS Linux release 7.6.1810 (Core)
CentOS Linux release 7.6.1810 (Core)
----------------
[root@host]# dmesg
[18573.314920] RDMA:INFO nvme_rdma_create_ctrl():6138 ctrl=ffff8a7db2a3d000
[18573.315038] opts->nr_io_queues =0
[18573.315075] nvme_rdma: enabling unsafe global rkey
[18573.325683] nvme nvme0: admin request pool allocated
[18573.362863] nvme_rdma: nvme_rdma_create_ctrl:6320   0: Queues size 0 total 0
[18573.362872] nvme nvme0: new ctrl: NQN "nqn.2014-08.org.nvmexpress.discovery", addr 192.168.6.10:4420
[18573.362878] RDMA:INFO nvme_rdma_create_ctrl():6417 ctrl=ffff8a7db2a3d000 &ctrl->ctrl=ffff8a7db2a3d198
[18573.363172] nvme0: We are not part of multipath group.
[18573.363450] RDMA:INFO nvme_rdma_free_queue():1425 nvme0 Queue id 0
[18573.364074] RDMA:INFO nvme_rdma_stop_and_free_queue():1438 nvme0 Queue id 0 inflight 0
[18573.364079] RDMA:INFO nvme_rdma_destroy_admin_queue():1864 qid 0 inflight 0
[18573.364083] RDMA:INFO nvme_rdma_destroy_admin_queue():1865 qid 0 inflight_rsvd_tags 0
[18573.368814] RDMA:INFO nvme_rdma_free_ctrl():1878 ctrl = ffff8a7db2a3d000

**Please Note: Issue is seen with nvme cli version 1.6 and greater**

keithbusch, 2019-10-10:
I think this was a kernel bug. Does this reproduce on the latest stable kernel (v5.3.5)?

keithbusch, 2019-11-21:
Any update? Pretty sure that was a kernel bug.

tabasumtamboli, 2019-11-22:
Its working fine, hence closing

closed 2019-11-22 by tabasumtamboli