linux-nvme/nvme-cli #595: Buffer size for can cause memory corruption under certain conditions.

Prgrmman, 2019-10-21:
https://github.com/linux-nvme/nvme-cli/blob/65192d2fbc4f47e6c6d8a93c350bc3d70ff9d730/nvme.c#L4279

In cases where the namespace is formatted to use use the extended LBA format, as specified by the nvme express standards v1.3, 

[image]

From the nvme format command, this would be specified with the "-m 1" option.

If the metadata is to be passed back in the same contiguous buffer as the data buffer, then it is insufficient to just use the "phys_sector_size" retrieved from an ioctl to the block device

Prgrmman, 2019-10-21:
This is causing memory corruption

Prgrmman, 2019-10-21:
This could be argued as "working as designed", and I suppose the user can specify the data size as larger to accommodate this extra data, but the code generally does not seem to support LBA formats with the behavior mentioned above.

Prgrmman, 2019-10-21:
I suppose that drives that are formatted in interleaved metadata mode should have users that are aware of this, and adjust the datasize accordingly. @keithbusch has answered this in #223.
And the code does support larger data sizes than what the block device supports. I suspect that this issue may be closed, but it would be nice to have confirmation.

keithbusch, 2019-10-21:
Yes, the code is "do what I say", not "do what I mean". If you say to use an inappropriate buffer, the code will do that.

It is possible to add more smarts into the program.

keithbusch, 2019-10-21:
Closing this as working as designed. We could add a new feature that queries the format and auto-fills the transfer size. That wasn't originally done because we wanted a 1:1 relationship for each cli command to nvme commands that you could watch in a protocol trace. We've diverged from that point for several other commands already, so I'm not really against doing it for the "read/write" commands since we still have the passthru commands to send exactly command if we want to isolate.

closed 2019-10-21 by keithbusch

Prgrmman, 2019-10-22:
Yes makes sense. Thanks for clearing this up