linux-nvme/nvme-cli #600: nvme format fail

GitYuanQu, 2019-10-30:
I try to erase NVMe SSD with secure mode, but it failed. detail as below:
============================
$nvme --version
nvme version 1.6
$nvme format -s 0 /dev/nvme0n1
[ 859.491564] nvme nvme0: Shutdown timeout set to 6 seconds
NVME Admin command error: INVALID_FORMAT: The LBA Format specified is not supported. This may be due to various conditions(410a)
============================
as I know, if I not specified the LBA through option --pi, LBA will use default value '0', so why the error message is show 'LBA Format specified is not supported'? actually I am not specified the LBA value. so could someone tell me what should I do to fix or workaround this issue? thanks.

keithbusch, 2019-10-30:
Maybe your device doesn't support SES (the '`-s`' option). Check '`nvme id-ctrl /dev/nvme0n1 | grep fna`' and see if bit 2 is set.

keithbusch, 2019-10-30:
Oh, you've -s 0, which should be the same as not setting -s at all... In that case, yeah, the program would use the previous lbaf the device is currently formatted with, which is supposed to always succeed. Sounds like you've a device specific, non-compliant firmware issue.

GitYuanQu, 2019-10-30:
em... I wrote a error message, the actual command I've execute is "nvme format /dev/nvme0n1 -s 1", and the result as upper failed output.

GitYuanQu, 2019-10-30:
I have check this device through command 'ls /dev/nvme*', it output as below:
==================
/dev/nvme0 /dev/nvme0n1 /dev/nvme0n1p1 /dev/nvme0n1p2 /dev/nvme0n1p3 /dev/nvme1 /dev/nvmen1
===================
I try the command "nvme id-ctrl /dev/nvme0n1 | grep fna", is shows as below:
=======================
fna   : 0x4
=======================

GitYuanQu, 2019-10-30:
I even try to detach the namespace from /dev/nvme0 via execute command 'nvme detach-ns /dev/nvme0 -n 1 -c 0' and delete namespace through 'delete-ns /dev/nvme0 -n 1', but it all failed.
it shows status is NS_INSUFFICIENT_CAPACITY. so base on upper message, could you help me to know what occured on my device and how could I format it success, or do you have any workaround?

GitYuanQu, 2019-10-30:
have someone could answer this? I am waitting...

keithbusch, 2019-10-30:
Your controller's reported FNA says it supports secure erase on format options, so it should work either way.

I'm not sure what detaching the namespace was supposed to accomplish though. There really are not very many controllers/subsystems that support this behavior.

It really sounds like your device is not behaving in a compliant manner. Does an '`nvme format /dev/nvme0n1 -l 0`' succeed? That's the simplest type of format you can do.

GitYuanQu, 2019-10-30:
unfortunatelly, I try to use command 'nvme format /dev/nvme0n1 -l 0/1/2/3', its all fails.

keithbusch, 2019-10-30:
You should contact your device vendor.

keithbusch, 2019-10-30:
closing as a device side issue

closed 2019-10-30 by keithbusch

benjaminrich, 2019-12-18:
I have the same issue. My device controller identifies as:

```
mn        : WDC PC SN520 SDAPNUW-256G-1002          
```

Shekelme, 2021-04-09:
The same issue with WDS200T3X0C-00SJG0. On several computers it is possible to format them, but on one (Asus Pro WS WRX80E-SAGE SE WIFI, AMD 3995WX) it is not.

keithbusch, 2021-04-09:
Without any vendor guidance, there doesn't appear to be anything we can do from the application side to unblock this operation from executing.

tomty89, 2021-05-07:
[Hope you don't mind me spreading the possible solution]
See https://github.com/linux-nvme/nvme-cli/issues/816#issuecomment-834586681

aggieNick02, 2024-06-21:
Please see my comment on this related issue [here](https://github.com/linux-nvme/nvme-cli/issues/84#issuecomment-2183511051). It covers what is happening, why the suspend/resume trick often works, and what you can do when it does not.