linux-nvme/nvme-cli #602: nvme read issue as picture attached

adeamlee, 2019-11-05:
nvme read issue as picture attached。

if the size （-z）less than32， it works well；
or it crash as picture attached

please check.

adeamlee, 2019-11-05:
nvme read /dev/nvme0n1 -s 0 -c 16 -z 128 -d read_0_2.bin

adeamlee, 2019-11-05:
picture can not be uploaded

adeamlee, 2019-11-05:
code as below for allocate memory buffer has issue, please check.

	if (posix_memalign(&buffer, getpagesize(), buffer_size)) {
		fprintf(stderr, "can not allocate io payload\n");
		err = ENOMEM;
		goto close_mfd;
	}

keithbusch, 2019-11-05:
The command you are requesting to send is nonsense, I'm not surprised it crashes. The device is going to overrun the 128 byte buffer you requested to allocate for 17 blocks.

closed 2019-11-05 by keithbusch

adeamlee, 2019-11-06:
thanks a lot, i check the issue, the size also should be a multiple of 8 for that our test pc is 64bit Windows10.

keithbusch, 2019-11-06:
Not just a multiple of 8. It should be a multiple of the block size, which is usually either 512 or 4096.