linux-nvme/nvme-cli #605: Protocol corresponding to 1.9

zb861359, 2019-11-15:
Hi，where can I get the protocol corresponding to 1.9 version？

keithbusch, 2019-11-15:
The versioning of this tool is independent of the specification.

zb861359, 2019-11-15:
@keithbusch 
em...So which specification can I refer to in terms of version 1.9?
The most latest version of the specification is 1.4  on the official website.

keithbusch, 2019-11-15:
Oh, always refer to the public spec latest, which is 1.4. We don't implement all the features from the spec, but we add them as devices we can test implement them.

zb861359, 2019-11-18:
@keithbusch 
what do you mean of " we add them as devices we can test implement them."?

keithbusch, 2019-11-18:
New features are added to the program when a device that also implements it exists. That way the code can actually be tested. If the nvme spec defines a feature that not a single nvme device implements, then we're not going to bother updating the tool for an unreachable feature.

zb861359, 2019-11-19:
Is it means that there are nvme devices first, and then nvme programs? Whether this approach will affect the update iteration speed of nvme devices?

keithbusch, 2019-11-19:
Yes. The vendors who make the devices have prototypes implementing spec features long before they're generally available, though, so programs are typically enabled ahead of the market.

What feature are you looking for anyway?

zb861359, 2019-11-19:
I am looking for features of zoned namespace

keithbusch, 2019-11-19:
> I am looking for features of zoned namespace

There is no ratified TP defining that, so such topics on public lists isn't appropriate. You should contact your nvme technical workgroup representative for such things.

closed 2019-11-19 by keithbusch