linux-nvme/nvme-cli #609: NVME namespace not ready for I/O, but crontoller is operational, and responds to admin commands

Prgrmman, 2019-11-19:
https://github.com/linux-nvme/nvme-cli/blob/44755ae6869ab2a9dc6ac976fb43f4f2d746336c/nvme-print.c#L2676

After power cycling a drive, I noticed that the drive was posted to the OS and would respond to inquiry commands. However, it would not respond to block device I/O or nvme passthru I/O and was giving NVME_SC_NS_NOT_READY.

How can a drive be considered "operational" but not respond to I/O commands?
Even the CSTS.RDY register was set to 1.

PurushGupta, 2019-11-19:
I would check the smart-log

nvme smart-log /dev/nvme

if it had any critical warning and also consult the user manual/reach out
to vendor

CSTS.RDY indicates the nvme interface ready, the Flash controller needs to
have media ready condition to serve IO

Prgrmman, 2019-11-19:
@PurushGupta Well, the thing is that namespace does become ready after a while and begins taking I/O. This takes about 20-30 seconds. 

Can this media condition be reliably checked as some primitive form of "Test Unit Ready"?

Prgrmman, 2019-11-19:
@PurushGupta this does not have anything to do with critical warnings being posted in the smart log

keithbusch, 2019-11-20:
This is a stupid trick some vendors play to say "look how fast my controller bus ready!" when it is not really ready. Some would even say this is not spec compliant without also including CRDT features...

closed 2019-11-20 by keithbusch

Prgrmman, 2019-11-20:
Thanks @keithbusch.

We confirmed with our vendor that they had explicitly modified their firmware at the request of another org.

Frustrating, but good to know that the nvme-cli code is getting it right.
Many thanks

keithbusch, 2019-11-20:
Yes, that's always the excuse. It is a short-sighted request considering they're going to have to waste time handling errors rather than waste time waiting for a really ready controller. No one has explained why that trade is useful to anyone.

Okisa123, 2019-11-20:
Keith, can you point what did you mean “Some would even say this is not spec compliant without also including CRDT features...”
Olga

> On Nov 19, 2019, at 5:27 PM, Keith Busch <notifications@github.com> wrote:
> 
> Some would even say this is not spec compliant without also including CRDT features...
>

keithbusch, 2019-11-20:
The host must wait a completely arbitrary time for a real ready before retrying a command, and that time to wait is going to be different for each vendor. That is unacceptable and can not be properly verified as a spec compliant.

CRDT removes any arbitrary wait time in a spec complaint manner. Please find TP 4033 on nvmexpress.org for more details.