linux-nvme/nvme-cli #612: nvme self-test-log outputs "Test is 0% complete and is still in progress."

courtneybanh, 2019-11-21:
Nvme-cli is reporting "Test is 0% complete and is still in progress" when there is no test running or the test has already fully completed.  The output would be more accurate if nvme-cli reported something like "No device self-test in progress".  nvme-cli does give correct information when a self-test is running, i.e., showing the percent complete.

`# nvme self-test-log /dev/nvme0
Test is 0% complete and is still in progress.
# nvme version
nvme version 1.9.122.gc509`

keithbusch, 2019-11-21:
oh, that is incorrect indeed. Must have slipped by me in review, i'll queue up a better decoding description.

closed 2019-11-21 by keithbusch

courtneybanh, 2019-11-21:
Will this change still show the correct message w/ percent completion if a device self-test is in progress?  Since this line has now been removed:
```
fprintf(stderr, "Test is %d%% complete and is still in progress.\n",
			self_test->crnt_dev_selftest_compln);
```
Would we still want to keep the above line and instead add a condition checking if the percentage is 0?  Otherwise it will always print the line below regardless of percentage, according to the latest change:
```
if (self_test->crnt_dev_selftest_oprn == 0) {
		fprintf(stderr, "No device self-test operation in progress\n");
		return;
	}
```
Also the log structure is currently not being printed in any format (normal, json, or binary).  Is the return statement at line 3531 preventing the rest of the function from executing?

keithbusch, 2019-11-21:
Based on my quick review of the spec, operation 0 means the log is empty.

courtneybanh, 2019-11-21:
The current device self-test operation field specifies the current action taken by the device self-test command.  Value of 0 indicates that no device self-test operation is in progress, not that the log is empty.  There may be no self-test running, but there could be existing self-test result data structures from previously completed operations.
[image]

keithbusch, 2019-11-21:
Okay, you're right. Looking through this logs reflection routine, I can't say it makes a whole lot of sense to me. A 'blame' says points to a commit from me, but that's only because I relocated the function while doing some house cleaning... I'll rewrite it to something more sane; I guess this log page hasn't been very well tested! Thanks for the notice.

courtneybanh, 2019-11-21:
Thanks @keithbusch!

keithbusch, 2019-11-22:
should be much better now. had to emulate a drive with these capabilities though.

filipporemonato, 2023-01-19:
Hi, I have this exact issue.
Installed nvmi-cli two days ago in the normal way
`sudo apt install nvme-cli`
and started a long test with
`sudo nvme device-self-test /dev/nvme0n1p2 -s 2h`
but after two days I still get "Test is 0% complete and is in progress" when running `sudo nvme self-test-log /dev/nvme0n1p2`.
Version is 
```
# nvme version
nvme version 1.9
```

igaw, 2023-01-19:
The current upstream self-test-log output is a bit different. This message doesn't exist in this form anymore.

filipporemonato, 2023-01-19:
Hei @igaw , thanks for the reply but I'm a bit confused by the "this message doesn't exist in this form anymore". I installed nvme-cli just two days ago... did I do a mistake by installing and got an older version? I'll double-check

igaw, 2023-01-19:
You are reporting against this upstream project. Though Ubuntu ships an older version thus you need to report Ubuntu. We are not fixing/maintaining downstream versions.