linux-nvme/nvme-cli #614: telemetry-log returns EIO (5) on successful completion

alexeyponomarev, 2019-11-22:
Hello, 

With Intel datacenter NVMe SSD starting with nvme-cli 1.9 the tool started to return exit code 5 on successful telemetry-log command:

> [root@localhost nvme-cli-**1.8**]# strace ./nvme telemetry-log /dev/nvme0 -o /tmp/telemetry.bin 2>&1 | tail -n 2
> exit_group(512)                         = ?
> +++ **exited with 0** +++

> [root@FM21V213-PCI35 nvme-cli-**1.9**]# strace ./nvme telemetry-log /dev/nvme0 -o /tmp/telemetry.bin 2>&1 | tail -n 2
> exit_group(5)                           = ?
> +++ **exited with 5** +++

Seems like the tool believes that EIO shall be returned because going out of **get_telemetry_log()** the err (and status) is 512 (0x200) - see gdb output attached - which after '& 0x700' and '>> 8' in nvme-status.c translates to SCT=2 and EIO return code as a result.

[nvme-cli-v18-gdb.txt](https://github.com/linux-nvme/nvme-cli/files/3878153/nvme-cli-v18-gdb.txt)
[nvme-cli-v19-gdb.txt](https://github.com/linux-nvme/nvme-cli/files/3878154/nvme-cli-v19-gdb.txt)

Verified that the command actually finishes successfully:
1. The resulting file is valid, no errors during execution in stderr;
2. PCIe trace capture of this command shows that each and every Get Log Page finished SCT=0, SC=0 (Generic Command Status + Successful Completion) in every Admin CQE packet;

Looks like both 1.8 and 1.9 end up with status=512, but with new functionality of 1.9 it results in EIO return code for the shell?

RevanthRajashekar, 2019-11-22:
This is because, the variable err contains value of number of bytes written. This has been fixed by https://github.com/linux-nvme/nvme-cli/pull/613
Do a git pull and it should fix it.

keithbusch, 2019-11-24:
Can we close this issue? Looks like the fix is already included.

closed 2019-11-25 by alexeyponomarev