linux-nvme/nvme-cli #620: Trying to find out how to format Crucial P1 to 4k sectors

sjmudd, 2019-12-03:
Not sure if here is the right place to ask. I have a couple of crucial P1 1 TB drives reported as follows:
```
$ sudo nvme list /dev/nvme0n1
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev  
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     XXXXXXXXXXXX         CT1000P1SSD8                             1           1.00  TB /   1.00  TB    512   B +  0 B   P3CR013 
$
```
but it uses 512 byte sectors. Documentation on the crucial.com website is not very good so I could not see how to format to 4k. I contacted support who said they did not support linux but they did say that using Windows I would be able to get 4k sector formatting. I just do not know how.

To format with 512-byte blocks this works:
```
$ sudo nvme format -l 0 /dev/nvme0n1
Success formatting namespace:1
$
```

and on a micron 9200 drive I had access to I use use `-l 1` to do this but trying on the P1 gives:

```
$ sudo nvme format -l 1 /dev/nvme0n1
NVME Admin command error:INVALID_FORMAT: The LBA Format specified is not supported. This may be due to various conditions(10a)
$ 
```
so I wondered if you might have any ideas on how to achieve this?  Any thoughts or pointers would be most appreciated.

keithbusch, 2019-12-03:
On Tue, Dec 03, 2019 at 02:09:40PM -0800, Simon J Mudd wrote:
> Not sure if here is the right place to ask. I have a couple of crucial P1 1 TB drives reported as follows:
> ```
> $ sudo nvme list /dev/nvme0n1
> Node             SN [...]

There's nothing here to inidicate what lba format index 0 means. We need
to see what it is first:

  # nvme id-ns /dev/nvme0n1
  ...
  lbaf  0 : ms:0   lbads:9  rp:0x2 (in use)
  lbaf  1 : ms:0   lbads:12 rp:0x1

In the example above, lba format index 1 corresponds to a 4k format with
an lbads of 12 (2^12 == 4k).

This is just an example for my particular drive, though. You will need
to check your drive's id-ns to see which lbaf to select in your format
command.

sjmudd, 2019-12-04:
@keithbusch thanks for responding. I see the following:
```
$ sudo nvme id-ns /dev/nvme0n1
NVME Identify Namespace 1:
nsze    : 0x74706db0
ncap    : 0x74706db0
nuse    : 0x74706db0
nsfeat  : 0
nlbaf   : 0
flbas   : 0
mc      : 0
dpc     : 0
dps     : 0
nmic    : 0
rescap  : 0
fpi     : 0x80
dlfeat  : 1
nawun   : 0
nawupf  : 0
nacwu   : 0
nabsn   : 0
nabo    : 0
nabspf  : 0
noiob   : 0
nvmcap  : 0
nsattr  : 0
nvmsetid: 0
anagrpid: 0
endgid  : 0
nguid   : 000000000000000100a07519e223f1d2
eui64   : 00a07519e223f1d2
lbaf  0 : ms:0   lbads:9  rp:0 (in use)
$
```
So only lbaf 0 is shown and 2^9 = 512 is all that is shown. So maybe here 512 blocks is all I can use?

sjmudd, 2019-12-04:
Note: the drive works. I was just under the impression that using it in 4k block mode was "better" as often that is the underlying size the drive uses.

However it is strange that the online support technician I spoke to said:

```
support (03/12/2019, 20:49:10): At this point we could suggest you to download storage executive. 
support (03/12/2019, 20:49:40): https://www.crucial.com/usa/en/support-storage-executive 
support (03/12/2019, 20:50:50): you will be able to format to to use 4k sectors. 
```

This is a Windows only product which does not help.

```
support (03/12/2019, 20:58:24): then Im sorry we do not have any updates for linux currently, I'll inform the concern team for future updates. 
```

So looks like I may be out of luck. I do hope they bump into this issue and help you guys make it possible to do this directly.

keithbusch, 2019-12-04:
The device doesn't support a 4k physical block. Sounds like they're just doing some hack to tell OS to use a 4k logical.

sjmudd, 2019-12-04:
Ok.  Thanks for your input.

closed 2019-12-04 by sjmudd