linux-nvme/nvme-cli #621: admin-passthru not passing data as expected

kellurs, 2019-12-04:
I am running the following command:
admin-passthru /dev/nvme1n1 -o 0xd0 -l 20 --cdw10=5 --cdw12=0xabcd1251 --cdw13=0xDAADBEEF -w -s

then I put an input string: qwopqwop

This is the output:
opcode       : d0
flags        : 00
rsvd1        : 0000
nsid         : 00000000
cdw2         : 00000000
cdw3         : 00000000
data_len     : 00000014
metadata_len : 00000000
addr         : 1e3c000
metadata     : 0
cdw10        : 00000005
cdw11        : 00000000
cdw12        : abcd1251
cdw13        : daadbeef
cdw14        : 00000000
cdw15        : 00000000
timeout_ms   : 00000000

However, on the PCIe analyzer when I pull the data I can see it is all zeros.

How can I use the data transfer on admin passthru? I set AVSCC in identify as per the NVMe 1.4 reference but it does not change the result.

keithbusch, 2019-12-04:
NVMe commands are supposed to encode the data direction in the opcode's lower two bits, and 0xd0 doesn't encode anything at all. I suspect the kernel thinks this is a data-in command and is bouncing what it thinks is read data through a zero buffer.

kellurs, 2019-12-04:
Yes, that was the issue, thanks. Would it be helpful to add a warning for cases where the read or write flag is set but does not agree with the opcode?

closed 2019-12-04 by kellurs