linux-nvme/nvme-cli #622: missing hostnqn

s9128122, 2019-12-04:
I got  
[21425.245948] nvme_fabrics: unknown parameter or missing value 'hostnqn=' in ctrl creation request
from dmesg. 
And I find /etc/nvme/hostnqn file is blank. Used "nvme gen-hostnqn" command, then it's show 
"gen-hostnqn" not supported. Install lib uuid and rebuild.

Could everyone help me to solve the problem?

OS： CentOS 7.7 1908  3.10.0-1062.4.3.el7.x86_64
nvme version：1.9.123.g972e

keithbusch, 2019-12-04:
Run 'make clean' in the nvmecli source first, then 'yum install libuuid'. Rebuild nvme with 'make' again and it should find the appropriate libraries to generate.

Alternatively, you should be able to define your own hostnqn at the command line with a '--hostnqn=myhostnqn' in your connect command.

s9128122, 2019-12-05:
Thank you very much, it solved.
After your indication, nvme gen-hostnqn works, then I run echo newhostnqn > /etc/nvme/hostnqn.
So I can find my nvme form another server by NVMe-oF.
Appreciate your help .

closed 2019-12-05 by s9128122

s9128122, 2019-12-06:
More note, not libuuid.
For CentOS：yum install libuuid-devel
For Ubuntu：apt-get install uuid-dev

jasonwee, 2024-04-10:
> For CentOS：yum install libuuid-devel

yes, works, thank you.