linux-nvme/nvme-cli #624: nvme: kato value is not honored on discover

bharatpotnuri, 2019-12-05:
Hi All,
I encountered this issue while trying to discover using linux host to and spdk target.
spdk target doesnt allow kato during discovery. looks like nvmecli is passing down some kato value irespective of -k option passed. I tried giving -k0 to make it work with spdk but I still see some value as kato at the target. Packet trace also shows 15000 as kato value when I deliberately gave -k 0. its the same if I give -k 30 assuming 30 secs kato.
I see this behavior with TOT nvmcli only. It works fine with v1.3 nvme-cli that came with distro.

looks like the below commit caused this behavior:
```
commit 6bdd3db066f9a9a53bca32d310a2ce77818ce4a7
Author: Hannes Reinecke <hare@suse.de>
Date:   Thu Aug 1 10:47:41 2019 +0200

    nvme-cli: mask out invalid options during discovery

    When compiling the option string in build_options() we should ensure
    to not add invalid options when doing a discovery. So add an additional
    option 'discovery' to build_options() to identify these cases.
```
I am not sure if it can be reverted so didnt send the patch.

bharatpotnuri, 2019-12-23:
@keithbusch Am I missing something here?

keithbusch, 2019-12-23:
I'm primarily development on pcie, i'll need to refresh the fabrics side to understand what's going on here.

keithbusch, 2019-12-23:
kernel defaults to a 5 kato, adds 10 "grace", so default kato is 15000. nothing to do with this tool.

the tool, however, won't allow a "0" kato value, and the kernel will warn if you do use such a value.

bharatpotnuri, 2019-12-26:
> 
> 
> kernel defaults to a 5 kato, adds 10 "grace", so default kato is 15000. nothing to do with this tool.
> 
> the tool, however, won't allow a "0" kato value, and the kernel will warn if you do use such a value.
Ok, I have riased an issue in SPDK-NVMF to support KATO. Closing this issue.
Thanks @keithbusch

closed 2019-12-26 by bharatpotnuri

keithbusch, 2019-12-26:
Well, i'm not necessarily saying what the kernel is doing is always correct either, but I'm not familiar enough with the fabrics protocol to know off the top of my head.

bharatpotnuri, 2019-12-27:
> 
> 
> Well, i'm not necessarily saying what the kernel is doing is always correct either, but I'm not familiar enough with the fabrics protocol to know off the top of my head.

fine. The nvmf protocol, as I understand, specifies discovery kato to be fixed value and thats why target doesnt accept kato. but recently there were couple of changes in kernel making target accept  kato. That is when it broke SPDK NVMF target. I observed the current issue when I tried working it areound by passing kato '0' which worked with older verions of nvme-cli.
This is fixed in SPDK by making spdk target accept kato during discovery. 
For reference https://github.com/spdk/spdk/issues/1089
I shall pursue this later and probably comeback with a fix.
Thanks for your time @keithbusch.