linux-nvme/nvme-cli #627: NVMe status: INVALID_OPCODE: The associated command opcode field in not valid(0x6001)

danarellano, 2019-12-09:
Version: nvme-1.9.143.g525a

`sudo nvme format -s1 /dev/nvme0n1`
`NVMe status: INVALID_OPCODE: The associated command opcode field in not valid(0x6001)`

`sudo nvme format -s1 /dev/nvme0n1`
`systemctl -i suspend`
`NVMe status: INVALID_OPCODE: The associated command opcode field in not valid(0x6001)`

Any suggestions?

keithbusch, 2019-12-09:
Your device is not correctly implementing the nvme protocol. There's nothing tooling can do about this, you can only complain to your vendor, or just purchase a more compliant drive.

keithbusch, 2019-12-09:
Eh, it could still be compliant since Format NVM is listed as an optional command. Check '`nvme id-ctrl | grep oacs`' and see if bit 1 is set to confirm the device supports the command. If it is not set, then the device is correctly announcing that it doesn't support any type of format, and your only recourse is still to contact your vendor if you want this command implemented.

danarellano, 2019-12-10:
`sudo nvme id-ctrl /dev/nvme0n1 | grep oacs`
result in:
`oacs         : 0x17`

0x **1** 7 
Does this confirm the devices supports the command?

keithbusch, 2019-12-10:
Yes, that says format is supported. Sounds like a firmware bug to me. Either it needs to unset the bit or we shouldn't see that error status.

closed 2019-12-10 by keithbusch

maximbaz, 2019-12-30:
If your drive is made by Samsung, there is a known bug that results in this error, the workaround is to suspend your computer first (systemctl suspend), then resume, then try formatting again.

deepakagrawal, 2020-07-18:
I also get 

```
sudo nvme id-ctrl /dev/nvme0n1 | grep oacs
result in:
oacs : 0x17

```

You mentioned `Either it needs to unset the bit or we shouldn't see that error status.` How do I do that? I tried suspending the system as well but no success.

deepakagrawal, 2020-07-18:
> `sudo nvme id-ctrl /dev/nvme0n1 | grep oacs`
> result in:
> `oacs : 0x17`
> 
> 0x **1** 7
> Does this confirm the devices supports the command?

Were you able to solve this problem @danarellano

keithbusch, 2020-07-18:
> You mentioned `Either it needs to unset the bit or we shouldn't see that error status.` How do I do that? I tried suspending the system as well but no success.

You don't do that. The vendor is supposed to do that because their implementation is broken.