linux-nvme/nvme-cli #628: modprobe: FATAL: Module nvme-tcp not found in directory /lib/modules/4.15.0-45-generic

Wangzihao18, 2019-12-10:
After `make` and `make install`
The commands `modprobe nvmet` , `nvme-rdma`, `nvmet-rdma`, `nvme-fc`  and `nvmet-fc` can run successfully.
But commands `modprobe nvme-tcp` and `modprobe nvmet-tcp` fails.
```
modprobe: FATAL: Module nvme-tcp not found in directory /lib/modules/4.15.0-45-generic
modprobe: FATAL: Module nvmet-tcp not found in directory /lib/modules/4.15.0-45-generic
```

Here are my envs
```
NAME="Ubuntu"
VERSION="18.04.2 LTS (Bionic Beaver)"
Linux ubuntu 4.15.0-45-generic #48-Ubuntu SMP Tue Jan 29 16:32:18 UTC 2019 aarch64 aarch64 aarch64 GNU/Linux
```

keithbusch, 2019-12-10:
This is the wrong forum for this request. This repo is only user soace tooling, but you're asking about a kernel module.

As far as nvme tcp support goes, that wasn't introduced to the kernel until version 5.0. We never 
back port new features and drivers in the community supported stable kernel, so you'll never see it in a 4.15 kernel from the stable tree.

But you're using a distro kernel, which doesn't necessarily abide by community stable. You'd have to file a feature request with your distro to see if they'd take on this. Ubuntu's tracker is at https://launchpad.net/ubuntu. I am skeptical they'd take on porting the nvme tcp driver back to a 4.15 base, though, they'd probably recommend using a newer HWE kernel or upgrading to something more recent like 19.04 or 19.10.

closed 2019-12-10 by keithbusch

Wangzihao18, 2019-12-10:
Thanks for your answer @keithbusch