linux-nvme/nvme-cli #629: attach-ns / detach-ns CONTROLLER_LIST_INVALID without specifying "-c"

alexeyponomarev, 2019-12-14:
Help sections for attach-ns / detach-ns show that controller list "-c" argument is optional, yet with nvme-cli 1.9 commands without specifying controller list manually fail:

[root@localhost nvme-cli-1.9]# ./nvme attach-ns /dev/nvme0 -n 1
**NVMe status: CONTROLLER_LIST_INVALID: The controller list provided is invalid(0x411c)**
[root@localhost nvme-cli-1.9]# dmesg
[ 3015.352997] nvme nvme0: [0000:5e:00.0 | 0 | 01] REQUEST *** : [QID: 00] opcode: 0x15, flags:00 cmdid: 0x16, nsid: 0x00000001, DW10-15 [0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000]
[ 3015.353080] nvme nvme0: [0000:5e:00.0 | 0 | 01] *** response: [QID: 00] result: 0x00000000, sq_head 0x10, sq_id 0x00, cid 0x16 __ status: **0x8239**

[root@localhost nvme-cli-1.9]# **./nvme attach-ns /dev/nvme0 -n 1 -c 0**
**attach-ns: Success, nsid:1**
[root@localhost nvme-cli-1.9]# dmesg
[ 3015.352997] nvme nvme0: [0000:5e:00.0 | 0 | 01] REQUEST *** : [QID: 00] opcode: 0x15, flags:00 cmdid: 0x16, nsid: 0x00000001, DW10-15 [0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000]
[ 3015.353080] nvme nvme0: [0000:5e:00.0 | 0 | 01] *** response: [QID: 00] result: 0x00000000, sq_head 0x10, sq_id 0x00, cid 0x16 __ status: 0x8239
[ 3023.095573] nvme nvme0: [0000:5e:00.0 | 0 | 01] REQUEST *** : [QID: 00] opcode: 0x15, flags:00 cmdid: 0x1A, nsid: 0x00000001, DW10-15 [0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000]
[ 3023.097941] nvme nvme0: [0000:5e:00.0 | 0 | 01] *** response: [QID: 00] result: 0x00000000, sq_head 0x11, sq_id 0x00, cid 0x1A __ status: **0x01**

keithbusch, 2019-12-14:
Your controller is returning that error, not the tool. The tool is just the messenger.

Spec says you can define a 0 length controller list, and your controller thinks otherwise. Not that it's useful to send this command with such a list (it should be a no-op), but your controller isn't spec compliant in this regard either.

alexeyponomarev, 2019-12-15:
@keithbusch agree with what you're saying, originally opened the issue since: 1) a few other tools mark this argument as mandatory (while others e.g. ISDCT also make it optional) and 2) tested this on several generations Intel DC PCIe SSDs and none supported that scenario.

Looking closer at the spec as you mentioned - indeed appears that it's up for the vendor how to handle this scenario, so it's right that nvme-cli marks the controller list as optional argument indeed.  Thanks for the feedback, closing the issue

closed 2019-12-15 by alexeyponomarev