linux-nvme/nvme-cli #636: RoCE/nvme-cli over Bonded Interfaces

rpondr, 2019-12-21:
Disclaimer, I have read bonding interfaces that use RoCE is unsupported by Mellanox and possibly the Linux inbox driver but I am looking for a workaround so that my SAN can maintain HA.

Everything works great without a bond but here is the behavior when the bond is introduced.

Setup:
- Dual port Connext X5 100g Card with OFED driver
- 10 NVME LUNS connected to kernel with nvme-cli
- GPFS heartbeat/backend is on the some interface/bond

Error with bond:
- Connect command failed, error wo/DNR bit: 7
- 4 out of 10 drives connect

keithbusch, 2019-12-22:
I think the nvme kernel mailing list may be more appropriate for this, linux-nvme@lists.infradead.org.

rpondr, 2019-12-23:
Thanks for pointing me in the right direction. From my own testing it does look like bonds are working. I even tested the port failover. My issue was connecting the drives before the bond was formed during boot.

I still have a related issue. Inconsistently I see the following DNR error on connect.

Connect command failed, error wo/DNR bit: 7

keithbusch, 2019-12-23:
DNR is in the status the target is returning. Are you trying to connect to a linux in-tree kernel target, or something else?

rpondr, 2019-12-23:
The target is [paviliondata](https://pavilion.io/)

The target has this error:
IBV_WC_RETRY_EXC_ERR (12) - Transport Retry Counter Exceeded: The local transport timeout retry counter was exceeded while trying to send this message.

I think mellanox is talking about this QP error in there article here:
https://community.mellanox.com/s/article/bonding-considerations-for-rdma-applications

closed 2022-01-19 by igaw