linux-nvme/nvme-cli #639: Overlapping range error(4114)

ayshtmr, 2020-01-02:
Can someone please point out the scenarios where one is likely to see `NVME Admin command error:OVERLAPPING_RANGE: This error is indicated if the firmware image has overlapping ranges(4114)` error when running `fw-download`. Is nvme-cli sensitive to image naming when with this command? 

I'm sorry but I could not find much details about this error in documentation.

keithbusch, 2020-01-02:
Spec says it may be returned if the granularity or alignment are not met, but it also says the controller may fail with Invalid Field in Command instead. Overlapping may indicate either an offset or transfer length is not aligned, or the offset + length overlaps with a previous firmware download command. The overlap detection is optional, and controllers are not required to implement checking range portions against previous commands, so YMMV on observing a controller return this status.

To hit an unaligned condition, you need a controller that has a a granularity (I've never seen one, but it'd show up in id-ctrl's 'fwug' field if it exists), then provide a `--xfer` value smaller than what that value decodes to (it's in 4k sized units).

To trigger an overlapping fw image, assuming the controller supports validating ranges, you could do something like run `nvme fw-download` twice with a non-zero --offset value the second time. A controller would more likely detect that as an Invalid Firmware Image if you'd try to commit it, though.

ayshtmr, 2020-01-03:
Thanks for the clarification. Is it possible clear up any previous fw-download on a drive so any subsequent fw-download attempts (with the same image) do not fail with the overlapping range error? I assume rebooting the drive could be one way but I'm wondering if there's something better.

keithbusch, 2020-01-03:
The spec only says what a controller "may" do in these situations, not "shall". The behavior is therefore vendor specific, so there is no generic way to answer your question beyond that. If you need further information on this scenario, you should contact your vendor.

closed 2020-01-03 by ayshtmr