linux-nvme/nvme-cli #641: 1.10 fails on install-man

frinnst, 2020-01-07:
/tmp/nvme-cli-1.10$ make PREFIX=/tmp install-man
make -C Documentation install-no-build
make[1]: Entering directory '/tmp/nvme-cli-1.10/Documentation'
install -d -m 755 /tmp/share/man/man1
install -m 644 nvme-lnvm-create.1 nvme-lnvm-init.1 nvme-wdc-clear-assert-dump.1 nvme-sanitize-log.1 nvme-help.1 nvme-sanitize.1 nvme-detach-ns.1 nvme-compare.1 nvme-attach-ns.1 nvme-wdc-drive-essentials.1 nvme-intel-temp-stats.1 nvme-security-send.1 nvme-wdc-get-crash-dump.1 nvme-wdc-vs-error-reason-identifier.1 nvme-smart-log.1 nvme-toshiba-clear-pcie-correctable-errors.1 nvme-show-hostnqn.1 nvme-intel-smart-log-add.1 nvme-list.1 nvme-lnvm-factory.1 nvme-effects-log.1 nvme-security-recv.1 nvme-netapp-smdevices.1 nvme-connect.1 nvme-huawei-id-ctrl.1 nvme-ns-descs.1 nvme-id-ctrl.1 nvme-device-self-test.1 nvme-ana-log.1 nvme-dir-receive.1 nvme-io-passthru.1 nvme-get-property.1 nvme-huawei-list.1 nvme-resv-acquire.1 nvme-reset.1 nvme-intel-internal-log.1 nvme-changed-ns-list-log.1 nvme-delete-ns.1 nvme-resv-register.1 nvme-virtium-save-smart-to-vtview-log.1 nvme-list-subsys.1 nvme-subsystem-reset.1 nvme-wdc-purge.1 nvme-lnvm-remove.1 nvme-disconnect.1 nvme-read.1 nvme-ns-rescan.1 nvme-lnvm-info.1 nvme-wdc-id-ctrl.1 nvme-wdc-vs-smart-add-log.1 nvme-gen-hostnqn.1 nvme-dir-send.1 nvme-get-feature.1 nvme-wdc-vs-nand-stats.1 nvme-disconnect-all.1 nvme-admin-passthru.1 nvme-set-property.1 nvme-lnvm-list.1 nvme-wdc-vs-telemetry-controller-option.1 nvme-get-log.1 nvme-create-ns.1 nvme-virtium-show-identify.1 nvme-wdc-purge-monitor.1 nvme-write-zeroes.1 nvme-dsm.1 nvme-wdc-vs-internal-log.1 nvme-intel-id-ctrl.1 nvme-wdc-clear-fw-activate-history.1 nvme-write.1 nvme-telemetry-log.1 nvme-flush.1 nvme-wdc-get-pfail-dump.1 nvme-intel-lat-stats.1 nvme-fw-download.1 nvme-lnvm-id-ns.1 nvme-resv-release.1 nvme-set-feature.1 nvme-wdc-vs-fw-activate-history.1 nvme-self-test-log.1 nvme-toshiba-vs-internal-log.1 nvme-connect-all.1 nvme-resv-report.1 nvme-fw-commit.1 nvme-get-ns-id.1 nvme-fw-log.1 nvme-wdc-cap-diag.1 nvme-write-uncor.1 nvme-discover.1 nvme-wdc-clear-pcie-correctable-errors.1 nvme-wdc-drive-resize.1 nvme-lnvm-diag-bbtbl.1 nvme-format.1 nvme-wdc-get-drive-status.1 nvme-dera-stat.1 nvme-toshiba-vs-smart-add-log.1 nvme-list-ctrl.1 nvme-lnvm-diag-set-bbtbl.1 nvme-list-ns.1 nvme-intel-market-name.1 nvme-wdc-drive-log.1 nvme-id-nvmset.1 nvme-netapp-ontapdevices.1 nvme-id-ns.1 nvme-show-regs.1 nvme-error-log.1 nvme-endurance-log.1 nvme.1 /tmp/share/man/man1
install: cannot stat 'nvme-wdc-vs-error-reason-identifier.1': No such file or directory
make[1]: *** [Makefile:135: install-no-build] Error 1
make[1]: Leaving directory '/tmp/nvme-cli-1.10/Documentation'
make: *** [Makefile:113: install-man] Error 2

frinnst, 2020-01-07:
Looks like the new tarball doesn't have pre-generated manpages for some things. Oversight? If not - please consider continuing shipping pre-generated man pages. It helps keep build-time deps down for us with source based distros :-)

keithbusch, 2020-01-07:
Yes, I missed the 'git add' for the wdc... I've pushed a fix, but it's unfortunately after the tag.

closed 2020-01-07 by frinnst